The-OpenROAD-Project/OpenROAD #506: Unable to install openroad on centos 7

honeysh08, 2020-09-12:
I followed the instruction given here and I ran the dockerfile. After the dockerfile is run successfully what are the steps that I am supposed to do in order to start using OpenROAD. After the completion, the build folder is still empty and hence, the cmake .. and the openroad these two commands are not working. I am completely new to this. Please provide help

maliberty, 2020-09-14:
I'm not sure where you mean by "here" in your description.

If you build with Docker then the final results will be stored in the docker image.  If you used a volume mount then you can look where you mounted it.  If not you'll need to run the image generated with 'docker run'.

maliberty, 2020-09-16:
Anything further or this ok to close?

honeysh08, 2020-09-16:
Yes, it has been solved. Thank you.

closed 2020-09-16 by honeysh08