The-OpenROAD-Project/OpenROAD #507: Internal error if generated clock is also a output pin

quentinw91, 2020-09-14:
Hi,
I have the following testcase, where a generated clock, called CLK_OUT, is also connected to an output pin.
[testcase_genclock2.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/5217470/testcase_genclock2.zip)

TritonCTS does not recognize it first as a clock tree (they are flops driven by CLK_OUT in the design) so i had to modify the cts.tcl script with the following lines:
```
set clock_nets [list]
lappend clock_nets "CLK"
lappend clock_nets "CLK_OUT"

# Run CTS
clock_tree_synthesis -lut_file "$::env(CTS_TECH_DIR)/lut.txt" \
                     -sol_list "$::env(CTS_TECH_DIR)/sol_list.txt" \
                     -root_buf "$::env(CTS_BUF_CELL)" \
                     -wire_unit 20 \
                     -clk_nets $clock_nets
```

Anyway, whether I have a clock tree inserted or not on CLK_OUT, i get an internal error while trying to do the final report:
 ```==========================================================================
report_checks -path_delay min
--------------------------------------------------------------------------
Startpoint: _27_ (rising edge-triggered flip-flop clocked by CLK)
Endpoint: _27_ (rising edge-triggered flip-flop clocked by CLK)
Path Group: CLK
Path Type: min

Fanout     Cap    Slew   Delay    Time   Description
-----------------------------------------------------------------------------
                          0.00    0.00   clock CLK (rise edge)
                          0.00    0.00   clock source latency
                  0.00    0.00    0.00 ^ CLK (in)
     1    3.71                           CLK (net)
                  0.00    0.00    0.00 ^ clkbuf_0_CLK/A (BUF_X4)
                  0.00    0.01    0.01 ^ clkbuf_0_CLK/Z (BUF_X4)
     2    2.25                           clknet_0_CLK (net)
                  0.00    0.00    0.01 ^ clkbuf_1_0_0_CLK/A (CLKBUF_X1)
                  0.01    0.03    0.05 ^ clkbuf_1_0_0_CLK/Z (CLKBUF_X1)
     3    3.22                           clknet_1_0_0_CLK (net)
                  0.01    0.00    0.05 ^ _27_/CK (DFFS_X2)
                  0.01    0.08    0.12 ^ _27_/QN (DFFS_X2)
     1    1.33                           clk_div.nxtCount[0] (net)
                  0.01    0.00    0.12 ^ _27_/D (DFFS_X2)
                                  0.12   data arrival time

                          0.00    0.00   clock CLK (rise edge)
                          0.00    0.00   clock source latency
                  0.00    0.00    0.00 ^ CLK (in)
     1    3.71                           CLK (net)
                  0.00    0.00    0.00 ^ clkbuf_0_CLK/A (BUF_X4)
                  0.00    0.01    0.01 ^ clkbuf_0_CLK/Z (BUF_X4)
     2    2.25                           clknet_0_CLK (net)
                  0.00    0.00    0.01 ^ clkbuf_1_0_0_CLK/A (CLKBUF_X1)
                  0.01    0.03    0.05 ^ clkbuf_1_0_0_CLK/Z (CLKBUF_X1)
     3    3.22                           clknet_1_0_0_CLK (net)
                  0.01    0.00    0.05 ^ _27_/CK (DFFS_X2)
                          0.00    0.05   clock reconvergence pessimism
                          0.01    0.06   library hold time
                                  0.06   data required time
-----------------------------------------------------------------------------
                                  0.06   data required time
                                 -0.12   data arrival time
-----------------------------------------------------------------------------
                                  0.07   slack (MET)

==========================================================================
report_checks -path_delay max
--------------------------------------------------------------------------
Startpoint: _26_ (rising edge-triggered flip-flop)
Endpoint: DATA_OUT (output port clocked by CLK)
Path Group: CLK
Path Type: max

Fanout     Cap    Slew   Delay    Time   Description
-----------------------------------------------------------------------------
                  0.02    0.00    0.00 ^ _26_/CK (DFFR_X1)
                  0.01    0.10    0.10 ^ _26_/Q (DFFR_X1)
     1    0.31                           DATA_OUT (net)
                  0.01    0.00    0.10 ^ DATA_OUT (out)
                                  0.10   data arrival time

                          2.00    2.00   clock CLK (rise edge)
                          0.00    2.00   clock network delay (propagated)
                          0.00    2.00   clock reconvergence pessimism
                         -1.00    1.00   output external delay
                                  1.00   data required time
-----------------------------------------------------------------------------
                                  1.00   data required time
                                 -0.10   data arrival time
-----------------------------------------------------------------------------
                                  0.90   slack (MET)

==========================================================================
report_checks -unconstrained
--------------------------------------------------------------------------
Startpoint: _26_ (rising edge-triggered flip-flop)
Endpoint: DATA_OUT (output port clocked by CLK)
Path Group: CLK
Path Type: max

Fanout     Cap    Slew   Delay    Time   Description
-----------------------------------------------------------------------------
                  0.02    0.00    0.00 ^ _26_/CK (DFFR_X1)
                  0.01    0.10    0.10 ^ _26_/Q (DFFR_X1)
     1    0.31                           DATA_OUT (net)
                  0.01    0.00    0.10 ^ DATA_OUT (out)
                                  0.10   data arrival time

                          2.00    2.00   clock CLK (rise edge)
                          0.00    2.00   clock network delay (propagated)
                          0.00    2.00   clock reconvergence pessimism
                         -1.00    1.00   output external delay
                                  1.00   data required time
-----------------------------------------------------------------------------
                                  1.00   data required time
                                 -0.10   data arrival time
-----------------------------------------------------------------------------
                                  0.90   slack (MET)

==========================================================================
report_tns
--------------------------------------------------------------------------
tns 0.00

==========================================================================
report_wns
--------------------------------------------------------------------------
wns 0.00

==========================================================================
report_check_types -max_slew -violators
--------------------------------------------------------------------------

==========================================================================
report_clock_skew
--------------------------------------------------------------------------
Error: Internal error in /OpenROAD/src/OpenSTA/search/Crpr.cc:78 missing prev paths.
```

jjcherry56, 2020-09-21:
The internal error is an opensta issue. Your zip file is only sorta kinda a test case, since it doesn't actually provide any of the files that cts.tcl reads. If you use the make cts_issue target in openroad-flow it makes a tarfile with everything necessary to reproduce the issue without building and running openroad-flow.

jjcherry56, 2020-09-21:
You will probably have a lot better luck if you define the generated clock on the output of the gate that drives the net. CLK_OUT is ambiguous because it references either a net or a port of that name. Generated clocks on nets are currently not supported by opensta and using the output port doesn't feed back to the loads connected to the net.

quentinw91, 2020-09-21:
I am not sure i am doing this correctly because i get an error:
```
bash-4.2$ make place
...
cp results/nangate45/testcase_genclock2/2_floorplan.sdc results/nangate45/testcase_genclock2/3_place.sdc
bash-4.2$ 
bash-4.2$ 
bash-4.2$ 
bash-4.2$ 
bash-4.2$ 
bash-4.2$ make cts_issue
# Creating runme.sh script
# Creating vars.sh/tcl script
sed: -e expression #1, char 0: no previous regular expression
make: *** [cts_issue] Error 1
```
Is there some documentation on this ?

quentinw91, 2020-09-21:
Anyway, yes it works if i define the generated_clock with the driving gate output pin.  But the problem i have is that the name of that cell is random...  So how could I write the sdc without having to run first the synthesis step ? Is there a way to control this (so we do not have to re-write the SDC each time the design is modified) ?

jjcherry56, 2020-09-21:
Some sdc kung foo makes it pretty trivial
set clk_gen_driver [get_pins -of [get_net clk_out] -filter "direction == output"]
create_generated_clock -name clk_gen -source clk -divide_by 2 $clk_gen_driver

I'd still like to know how to tickle that internal error tho.
I stay as far away from openroad-flow as I possibly can because it is such a nightmare to deal with.
As far as I am concerned it only obfuscates the openroad tcl scripts with make/shell scripts.
I think the main piece I am missing is the DEF from the output of cts. I can cobble together the
rest of the files from the nangate files we have.

quentinw91, 2020-09-22:
[20200921.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/5259520/20200921.zip)

jjcherry56, 2020-09-23:
thanks. fixed in opensta 9c8d9569 report_clock_skew internal error when target register is unclocked

closed 2020-09-23 by jjcherry56

jayeshec12, 2023-12-08:
Hello, how to tackle generated clock in SDC file??