The-OpenROAD-Project/OpenROAD #512: stdc++fs isn't needed on some systems

yurivict, 2020-09-15:
FreeBSD doesn't have such library, for example.

This comment https://stackoverflow.com/questions/54290254/problem-adding-stdfilesystem-to-cmake-project contains an advise with one way of fixing this.

OS: FreeBSD 12.2
clang-10

maliberty, 2020-09-19:
It is part of the gcc distribution.  If you have a version of gcc that supports c++17 then the library will be found with the compiler.

Note that we are not building on or generally supporting FreeBSD at this time.

yurivict, 2020-09-19:
I use the clang compiler. Clang doesn't have ```stdc++fs```.

Code like this can help to detect that this library is needed and add it to the list of libraries:
```
try_compile(NEED_FS_LIB "${CMAKE_BINARY_DIR}/temp"  "${CMAKE_SOURCE_DIR}/tests/has_filesystem.cc" 
            CMAKE_FLAGS -DCMAKE_CXX_STANDARD=17 -DCMAKE_CXX_STANDARD_REQUIRED=ON
            LINK_LIBRARIES stdc++fs)
if (NEED_FS_LIB)
  set(LIBS ${LIBS} stdc++fs)
else()
```

maliberty, 2020-09-22:
We just decided to drop the use of the filesystem APIs so this is now moot.

closed 2020-09-22 by maliberty