The-OpenROAD-Project/OpenROAD #513: [ICeWall] Invalid "FIXED" nets in the defok files

ax3ghazy, 2020-09-17:
These files have a "FIXED" attribute attached to some of the nets, which doesn't conform to the DEF standard. I am not sure yet how those were output by defout though.

Some examples:
https://github.com/The-OpenROAD-Project/OpenROAD/blob/41a93c7cfbdb35241553540897618108c4a8ccff/src/ICeWall/test/soc_bsg_black_parrot_nangate45.defok#L3879

https://github.com/The-OpenROAD-Project/OpenROAD/blob/openroad/src/ICeWall/test/soc_bsg_black_parrot_nangate45.flipchip.defok#L4030

Also, it's worth noting that there are a couple of "empty" nets (e.g., https://github.com/The-OpenROAD-Project/OpenROAD/blob/openroad/src/ICeWall/test/soc_bsg_black_parrot_nangate45.defok#L3860-L3861).

Colin-Holehouse, 2020-09-17:
The DEF in these files maches the DEF written out by OpenROAD to confirm the testcase is running as expected.
@maliberty - can you take a look at this from the point of view of the DEF writer

maliberty, 2020-09-18:
@Colin-Holehouse this looks like a ICeWall issue where it is created a fixed special wire but not assigning any shapes to it.

maliberty, 2021-02-24:
@ax3ghazy is this resolved?

closed 2021-04-20 by maliberty