The-OpenROAD-Project/OpenROAD #514: build faiils

jjts, 2020-09-22:
Running the install script (Docker) yields

 Cannot open: http://downloads.sourceforge.net/ltp/lcov-1.14-1.noarch.rpm. Skipping.
Error: Nothing to do
The command '/bin/sh -c yum install -y http://downloads.sourceforge.net/ltp/lcov-1.14-1.noarch.rpm' returned a non-zero code: 1

and exist

maliberty, 2020-09-22:
@vvbandeira would you take a look

vvbandeira, 2020-09-22:
Hi @jjts, could you provide the following info so I can try to reproduce the issue?

- OpenROAD git sha
- Docker version
- Command you ran on your shell

Also, this error always occurs? Or is it intermittent?

jjts, 2020-09-22:
Hi @vvbandeira 

I installed docker and then ran your install script build_openroad.sh

To test intermittence I ran it now a second time and it completed. However, I have no binaries in the build directories... (?)

Removing intermediate container b01e9b684755
 ---> f379aab553f2
Successfully built f379aab553f2
Successfully tagged openroad/flow:latest

Docker version 18.09.7, build 2d0083d

git log 
commit 9b503e51e99f2e039e23245bdfc0517a2bb04e4e (HEAD -> master, origin/master, origin/HEAD)
Author: Tom Spyrou <46945466+tspyrou@users.noreply.github.com>
Date:   Wed Aug 19 13:52:28 2020 -0700

    uprev yosys to master head

vvbandeira, 2020-09-22:
@jjts I believe you opened the issue for the wrong repository.
The commit you gave me is from the repo https://github.com/The-OpenROAD-Project/OpenROAD-flow-public
Also, the repo you opened the issue does not have a build_openroad.sh

Finally, the flow repo is currently archived (so I belive it is not to be used at this moment). @maliberty can you confirm that the flow repo is frozen for the moment?

maliberty, 2020-09-22:
Flow is currently archived but we should have a resolution soon.  Is a fix in flow required here?

rovinski, 2020-09-22:
@jjts See the [readme](https://github.com/The-OpenROAD-Project/OpenROAD-flow#option-1-docker), Docker build, step 3. If you build using docker, it creates a docker image. You need to build using option 2 to get the binaries in your local file system.

jjts, 2020-09-22:
Thanks, the Docker image actually runs. But I am confused. I was just
following the instructions in

https://openroad.readthedocs.io/en/latest/user/getting-started.html

So do you recommend Docker or not?

If it installs natively as in
https://github.com/The-OpenROAD-Project/OpenROAD it would be quite nice.
There does not seem to be many dependencies....

Com os melhores cumprimentos / Best regards,

Jose T. de Sousa

rovinski, 2020-09-22:
Those docs are very outdated. They also shouldn't even be part of this repository.

If you build with docker, it builds inside a docker container and makes a docker image with the binaries. You have to create a docker container to run the tools inside of (i.e. step 3 of the readme).
If you build locally, it builds in the local OS and filesystem.
It depends purely on how you want to run the tools.

jjcherry56, 2021-01-09:
See openroad/README.md for openroad build instructions.
As rovinski says, openroad.readthedocs.io is not maintained and is quite obsolete.

closed 2021-01-09 by jjcherry56