The-OpenROAD-Project/OpenROAD #515: Build failure: OpenDB/src/zutil/misc_functions.cpp:237:3: error: no matching function for call to 'time'

yurivict, 2020-09-22:
```
/wrkdirs/usr/ports/cad/openroad/work/OpenROAD-0.9.0-beta-637-gc3706485/src/OpenDB/src/zutil/misc_functions.cpp:237:3: error: no matching function for call to 'time'
  time(&wtn);
  ^~~~
/usr/include/time.h:160:8: note: candidate function not viable: no known conversion from 'long *' to 'time_t *' (aka 'int *') for 1st argument
time_t time(time_t *);
       ^
```

The code above isn't in the latest revision of OpenDB.

You might want to update git module's revision in OpenROAD.

maliberty, 2020-09-22:
It would be helpful if you said which branch you are discussing.  I see that openroad branch (our development branch) is pointed at the head.

yurivict, 2020-09-22:
I use 0.9.0-beta-637-gc3706485 - the current master branch.

osamahammad21, 2020-10-12:
@yurivict can you specify which OS are you working on?

yurivict, 2020-10-12:
> can you specify which OS are you working on?

FreeBSD 12.2

closed 2021-04-20 by maliberty