The-OpenROAD-Project/OpenROAD #519: def files missing

honeysh08, 2020-09-29:
so, I followed the steps given in openROAD-flow-public and installed it successfully. After running the make command in flow directory, following error occured. How do I resolve this?

mkdir -p ./results/nangate45/gcd ./logs/nangate45/gcd ./reports/nangate45/gcd
cp results/nangate45/gcd/1_1_yosys.v results/nangate45/gcd/1_synth.v
mkdir -p ./results/nangate45/gcd ./logs/nangate45/gcd ./reports/nangate45/gcd
cp designs/nangate45/gcd/constraint.sdc results/nangate45/gcd/1_synth.sdc
(/usr/bin/time -f "%Eelapsed %PCPU %MmemKB" openroad -no_init -exit ./scripts/floorplan.tcl) 2>&1 | tee ./logs/nangate45/gcd/2_1_floorplan.log
openroad: error while loading shared libraries: libQt5Widgets.so.5: cannot open shared object file: No such file or directory
0:00.01elapsed 0%CPU 224memKB
make: *** [results/nangate45/gcd/2_1_floorplan.def] Error 127

maliberty, 2020-09-29:
You need to add the directory containing libQt5Widgets.so.5 to your LD_LIBRARY_PATH.

honeysh08, 2020-09-29:
I am unable to find it, it seems to be missing

maliberty, 2020-09-29:
If it were missing then openroad would omit it.  On my ubuntu machine I see it in /usr/lib/x86_64-linux-gnu/

Try: find /usr -name libQt5Widgets.so.5 -print

honeysh08, 2020-09-30:
I am working on CentOS 7 and I tried to follow what you said, but I still couldn't find libQt5Widgets.so.5. Is there a way to install it?

honeysh08, 2020-09-30:
I found libQt5Widgets.so.5 at this location /usr/lib64 on my centOS 7. Now, what must be done?

maliberty, 2020-09-30:
sudo yum install -y qt5-qtbase-devel

honeysh08, 2020-10-01:
thanks, it worked.

closed 2020-10-01 by honeysh08