The-OpenROAD-Project/OpenROAD #521: Issues when cloning this github to directory

arelysn, 2020-10-01:
i am not well versed with github, and am not sure if this is an issue on my end or within the repo even after a few online searches for similar problems 

i am following the README to install, and i am getting the following error when cloning the **openroad-flow-public** to my directory using the command below from the README

> git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD-flow-public.git

Error:

> error: invalid path 'src/FastRoute/third_party/pdrev/src/aux.h'
> Submodule path 'tools/TritonRoute': checked out '83cf3047e9a401889b710ee54397d8b269b9f516'
> Submodule path 'tools/yosys': checked out 'ff4ca9ddc3fe3a0ef93feb295bf135ebff03b3df'
> Unable to checkout '07a33a2f65f3af21c1962322b6a4de0bec557438' in submodule path 'tools/OpenROAD'

Cloning seems okay. the checkout of OpenROAD is where there is issue.
Any help would be appreciated.

for further information. i have attempted the clone with github desktop and using git clone --recurse-submodules to no success and get the same error

arelysn, 2020-10-01:
@eder-matheus @jjcherry56

jjcherry56, 2020-10-01:
Since I can checkout both the master and openroad branches of the openroad repo this is not an openroad repo issue. It is an openroad-flow issue.

arelysn, 2020-10-01:
Thanks for your response.
When i try your third command
> `git checkout openroad`

i still get the below error 

> error: invalid path 'src/FastRoute/src/pdrev/src/aux.h'

Im going to try on a different machine. But, this should work on windows, correct? i would like to ultimately work directly on my machine.

maliberty, 2020-10-01:
As I recall windows doesn't like files named aux.  From https://docs.microsoft.com/en-us/windows/win32/fileio/naming-a-file

Do not use the following reserved names for the name of a file:

CON, PRN, **AUX**, NUL, COM1, COM2, COM3, COM4, COM5, COM6, COM7, COM8, COM9, LPT1, LPT2, LPT3, LPT4, LPT5, LPT6, LPT7, LPT8, and LPT9.

Thank DOS.

We should just rename the file to avoid this issue.

eder-matheus, 2020-10-01:
@maliberty I will update it now

maliberty, 2020-10-01:
Btw, nobody has ported openroad to windows, or plans to, so I expect you'll have a lot more trouble after this.  I suggest using WSL2  on windows to run ubuntu (that's what I do).

arelysn, 2020-10-01:
> Btw, nobody has ported openroad to windows, or plans to, so I expect you'll have a lot more trouble after this. I suggest using WSL2 on windows to run ubuntu (that's what I do).

Thanks i will look that up. I did not see any os recommendations in the readme or docs, but that makes sense.

closed 2020-10-01 by arelysn