The-OpenROAD-Project/OpenROAD #523: facing a error while run the RTL to GDS II for GCD file (error is installing the klayout)

Pratiknavadiya-12, 2020-10-06:
I am getting the following error even I install the QT

Project ERROR: Unknown module(s) in QT: designer multimedia multimediawidgets xmlpatterns svg

Running build (gmake all) ..
gmake: *** No targets specified and no makefile found. Stop.

How to resolve this? please help me.

maliberty, 2020-10-06:
Try sudo apt-get install qtmultimedia5-dev libqt5multimediawidgets5 libqt5multimedia5-plugins libqt5multimedia5

(from https://stackoverflow.com/questions/26948466/project-error-unknown-modules-in-qt-multimedia)

pratiknavadiya1, 2020-10-08:
Sir i had tried this command but it was not help me again same erorr was shown. So please help me and give another solution.

pratiknavadiya1, 2020-10-08:
Sir it shows the no package qtmultimedia5-dev libqt5multimediawidgets5 libqt5multimedia5-plugins libqt5multimedia5 avilable..
Error: Nothing to do

maliberty, 2020-10-08:
What OS are you using?

Pratiknavadiya-12, 2020-10-09:
I am using CentOs 7.  but I am also using the command yum install qtmultimedia5-dev libqt5multimediawidgets5 libqt5multimedia5-plugins libqt5multimedia5. 
but it did not help me. So please help me and give another solution.

maliberty, 2021-02-24:
Sorry this got dropped a while ago.  Can you try it in Docker?

closed 2021-04-20 by maliberty