The-OpenROAD-Project/OpenROAD #527: [FastRoute]: Repair antenna flow places diodes outside of rows. (Placed in rows check fails)

agorararmard, 2020-10-20:
In some (rare) cases, FastRoute places the diodes outside of the row causing the `placed in rows check` to fail. Interestingly enough, the design continues normally and TritonRoute doesn't complain (finishes DRC clean). But, of course, LVS reports mismatches. Here is an example:
[image]

And here is how to replicate it. 
[FR_row_check_testcase.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/5409215/FR_row_check_testcase.tar.gz)

eder-matheus, 2020-10-22:
@agorararmard In your testcase, both DEF files already has diodes inserted on it. Could you send the DEF file before diode insertion?

agorararmard, 2020-10-23:
@eder-matheus: that was a design that had diodes inserted but still had antenna violations, and I used FR to insert the new diodes.

Anyway, here is a design that didn't have any diodes, and then FR inserted the diodes and one of them was placed outside of the row `ANTENNA_1247`:
[FR_row_check_testcase.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/5429641/FR_row_check_testcase.tar.gz)

eder-matheus, 2020-11-07:
@agorararmard My last commit in openroad branch fix this issue. I've tested locally and diode cells are being correctly placed (verified with commercial tool)

agorararmard, 2020-11-16:
@eder-matheus: I checked a couple of test cases and the issue is gone. So, I'll close it. Thank you very much for your help!

closed 2020-11-16 by agorararmard