The-OpenROAD-Project/OpenROAD #534: verilog2def chokes on valid verilog syntax for assignement

smunaut, 2020-11-28:
`user_project_wrapper.synthesis.v, line 136 syntax error, unexpected '{', expecting ID.`

Verilog in question is :
```
assign { io_oeb[37:27], io_oeb[10:0] } = { one, one, one, one, one, one, one, one, one, one, one, one, one, one, one, one, one, one, one, one, one, one };
```

Which is perfectly valid and emitted by `yosys`.

jean-m-cyr, 2020-11-28:
Here's another where verilog front end is happy, but backend not so much!

```
Error: /project/openlane/user_project_wrapper/runs/user_project_wrapper/results/synthesis/user_project_wr
apper.synthesis.v, line 4 syntax error, unexpected REG.
Error: /project/openlane/user_project_wrapper/runs/user_project_wrapper/results/synthesis/user_project_wr
apper.synthesis.v, line 5 syntax error, unexpected REG.
Error: /project/openlane/user_project_wrapper/runs/user_project_wrapper/results/synthesis/user_project_wr

```
```
/* Generated by Yosys 0.9+3621 (git sha1 84e9fa7, gcc 8.3.1 -fPIC -Os) */

module \$paramod$86ee6077fd23a04b01bd50a3e77717e9daf7577f\sha3_256_miner_regs (clk, reset, read, write, a
ddr, wdata, la_write, la_input, ready, rdata, header_o, difficulty_o, start_nonce_o, control_o, solution_
i, status_i);
  reg \initial  = 0;
  reg [31:0] _00_;
  reg [31:0] _01_;
  reg [31:0] _02_;
  reg [31:0] _03_;
  reg [4:0] _04_;

```
See https://github.com/miscellaneousbits/caravel_sha3_256_crypto_miner.git branch miner, user_project_wrapper, to reproduce.

maliberty, 2020-11-30:
There is no verilog2def command in openroad.  Can you package a standalone test case with the verilog in question and the necessary openroad commands to reproduce it?

smunaut, 2020-12-01:
Ah yeah sorry, this is when doing a simple `read_verilog`

For instance : `test.v` : 

```verilog
module test (
	input wire [1:0] a,
	output wire b,
	output wire c
);

	assign {b,c} = a;

endmodule
```

And then : 
```
~$ openroad 
OpenROAD 0.9.0 b29a41e3b6
This program is licensed under the BSD-3 license. See the LICENSE file for details. 
Components of this program may be licensed under more restrictive licenses which must be honored.
Error: cannot open '/home/tnt/.openroad'.
% read_verilog /tmp/test.v
Error: /tmp/test.v, line 7 syntax error, unexpected '{', expecting ID.
```

jjcherry56, 2020-12-01:
see [https://github.com/The-OpenROAD-Project/OpenSTA/issues/31](url) for a work around

smunaut, 2020-12-01:
`splitnet -ports` was already present because it turns out it apparently doesn't cover everything.
However `yosys` just added `-simple-lhs` option to `write_verilog` to avoid this which does work around the issue.

It still remains that it'd be good if the verilog parser here actually supported all valid verilog syntax.

jjcherry56, 2021-01-20:
fixed in fd15d838 that integrates opensta commit c13383fb issue#31 verilog concat assign

closed 2021-01-20 by jjcherry56