The-OpenROAD-Project/OpenROAD #535: openroad read_verilog runs out of memory

koeaton, 2020-12-01:
I'm trying to synthesize a large open source core but run into issues with the "openroad read_verilog" command.

The verilog is read by yosys with no issue but when running the floorplan.tcl in the OpenROAD flow, my machine runs out of RAM running the "read_verilog" command.

The logs look like this and then crash after a while when the memory runs out:
```
OpenROAD 0.9.0 39a74cb5ad
This program is licensed under the BSD-3 license. See the LICENSE file for details. 
Components of the program may be licensed under more restrictive licenses which must be honored.
Notice 0: Reading LEF file:  ./platforms/nangate45/lef/NangateOpenCellLibrary.tech.lef
Notice 0:     Created 22 technology layers
Notice 0:     Created 27 technology vias
Notice 0: Finished LEF file:  ./platforms/nangate45/lef/NangateOpenCellLibrary.tech.lef
Notice 0: Reading LEF file:  ./platforms/nangate45/lef/NangateOpenCellLibrary.macro.mod.lef
Notice 0:     Created 134 library cells
Notice 0: Finished LEF file:  ./platforms/nangate45/lef/NangateOpenCellLibrary.macro.mod.lef 
```
The machine I'm using has 32 gb of RAM. Is there a way to make sure my machine doesn't run out of memory when executing this command? 

Thanks in advance.

maliberty, 2020-12-01:
Can you provide the test case?

By "running out of ram" do you mean the machine is starting to swap a lot or some failure happens?

koeaton, 2020-12-01:
Can you elaborate what you mean by a test case? Sorry, I'm new to a lot of this stuff.

And by running out of ram, I mean that a failure happens and the process is killed. When I check dmesg, it says "Out of memory: Kill process 12973 (openroad) score 873 or sacrifice child."
Now that you mention swap, my machine seems to not be swapping at all which might be part of the issue.

maliberty, 2020-12-01:
Would you provide a tar file with the tcl script you are running in openroad and any files it references (eg the verilog).

Perhaps you lack swap space.  Try "swapon -s" to see how you have available.

koeaton, 2020-12-01:
Apparently, I had 0 swap space. I added some space and that fixed the issue. 

Thanks for your help!

closed 2020-12-01 by koeaton