The-OpenROAD-Project/OpenROAD #538: handling file names with a space character fails for some procs

juspertor, 2020-12-21:
I found some issues in OpenRoad dealing with file-paths containing a space character. These issues can be fixed by changing 4 lines of code in 3 files:

file: src/OpenRoad.tcl

change line 49 from
set filename [file nativename $args]
to
set filename [lindex [file nativename $args] 0]

change line
set filename [file nativename $args]
to
set filename [lindex [file nativename $args] 0]

file: src/pdngen/src/PdnGen.tcl

change line 95 from
set config_file $args
to
set config_file [lindex $args 0]

file: src/OpenSTA/tcl/Liberty.tcl
change line 29 from
set filename [file nativename $args]
to
set filename [lindex [file nativename $args] 0]

$args is a list with one element in that case and the subsequent [file exists $filename] always returns false in that case for a list. So nothing is cut off. The solution with lindex is already used for read_sdc.

jjcherry56, 2020-12-22:
I am much more interested in seeing an test case or at the very least an example of the command and arguments that fail than I am in any proposed fix (which do not look correct).

juspertor, 2020-12-22:
Here is the test case. The folder "OpenROAD" contains the fixed version of OpenROAD while the folder "OpenROAD_20201219" contains the original version: 

[thies@localhost ~]$ mkdir test\ folder
[thies@localhost ~]$ cd test\ folder/
[thies@localhost test folder]$ touch test.lef
[thies@localhost test folder]$ cd ..      
[thies@localhost ~]$ OpenROAD_20201219/build/src/openroad 
OpenROAD 0.9.0 4007da0f4b
This program is licensed under the BSD-3 license. See the LICENSE file for details. 
Components of this program may be licensed under more restrictive licenses which must be honored.
% read_lef "/home/thies/test folder/test.lef"
Error: {/home/thies/test folder/test.lef} does not exist.
% ^C
[thies@localhost ~]$ OpenROAD/build/src/openroad 
OpenROAD 0.9.0 ce30b19496
This program is licensed under the BSD-3 license. See the LICENSE file for details. 
Components of this program may be licensed under more restrictive licenses which must be honored.
% read_lef "/home/thies/test folder/test.lef"
Notice 0: Reading LEF file:  /home/thies/test folder/test.lef
Notice 0: Warning: WARNING (LEFPARS-2001): No VERSION statement found, using the default value 5.8. See file /home/thies/test folder/test.lef at line 1.

Notice 0: Finished LEF file:  /home/thies/test folder/test.lef
%

jjcherry56, 2020-12-23:
Fixed in 0555f6e0 cmd filename args with spaces (not in the public repo yet).
Your fixes are close; the lindex should be on $args, not the file nativename.
set filename [file nativename  [lindex $args 0]]

juspertor, 2021-01-11:
The provided patch fixes the problem only partly. The file pdngen.tcl was not changed. So the problem still exists when reading the .pdn file:

file: src/pdngen/src/PdnGen.tcl

change line 95 from
set config_file $args
to
set config_file [lindex $args 0]

maliberty, 2021-01-11:
@Colin-Holehouse can you take a look at the pdngen issue

jjcherry56, 2021-01-19:
fixed in 
7e16f8e2 issue#538 pdngen file name with space

closed 2021-01-19 by jjcherry56