The-OpenROAD-Project/OpenROAD #542: Bad routability result in DAC2012 benchmarks

Kangkang-T, 2020-12-21:
I got previous versions before which was supported routability option and bookshelf format. Global routing is performed by the official global router NCTU-GR,  sHPWL is evaluated by the official DAC-2012 contest evaluation scripts, and detail placement is performed by NTUplace4h. I set target density 0.9 , superlinear inflation ratio 2.33(default), max inflation ratio 2.5(default), which were same as your paper “RePlAce: Advancing Solution Quality and Routability Validation in Global Placement” mentioned. But I got a worse RC result in DAC2012 benchmarks. For example, I got RC=433.97 after first inflation and RC= 215.11 after detail placement(10th inflation later) in superblue12. But it's RC =211,73 and RC = 102.43 in your paper. My result was too bad. was there something I made mistake or some options I forgot to set?
Thanks you.

mgwoo, 2020-12-28:
Sorry, We will not support the academic bookshelf benchmark anymore.

vijayank88, 2023-06-14:
Closing this as per @mgwoo reply. Re-open if needed.

closed 2023-06-14 by vijayank88