The-OpenROAD-Project/OpenROAD #543: Segfaults in the MuxTest*.v series of tests

oharboe, 2020-12-21:
Check out the latest alpha-release master and run:

```
#  make DESIGN_CONFIG=`pwd`/designs/tiny-tests.mk DESIGN_NAME=MuxTest_width_1_inputs_1_outputs_1_pipeline_0
[deleted]
INFO: Total 0 worst path!
/bin/bash: line 17:  2214 Segmentation fault      (core dumped) RePlAce -bmflag etc -lef ./objects/nangate45/MuxTest_width_1_inputs_1_outputs_1_pipeline_0/merged_padded.lef -def ./results/nangate45/MuxTest_width_1_inputs_1_outputs_1_pipeline_0/2_floorplan.def -verilog ./results/nangate45/MuxTest_width_1_inputs_1_outputs_1_pipeline_0/2_floorplan.v -lib ./platforms/nangate45/NangateOpenCellLibrary_typical.lib -sdc ./results/nangate45/MuxTest_width_1_inputs_1_outputs_1_pipeline_0/2_floorplan.sdc -output ./results/nangate45/MuxTest_width_1_inputs_1_outputs_1_pipeline_0/replace -t 1 -timing -resPerMicron 1.59 -capPerMicron 0.235146e-12 -skipIP -plot -experi output -den 0.509 -initCoef 0.00002 -onlyGP 2>&1
      2215 Done                    | tee ./logs/nangate45/MuxTest_width_1_inputs_1_outputs_1_pipeline_0/3_1_RePlAce.log

```

maliberty, 2021-01-13:
fixed - @tspyrou to close on next push.

oharboe, 2021-01-14:
tip: if you write "fixes #543" in the commit comment or in the pull request, the issue  will  autoclose on merge.

maliberty, 2021-01-14:
thanks.  Does that work when you merge from one repo to another?

oharboe, 2021-01-14:
Good point.

If you by that mean when you merge a pull request from a fork into the main repository, then yes.

We're using a mono-repository (using git subtree, completely transparent to users without git submodule hell), so we don't run into all these cross-repository problems.

maliberty, 2021-01-14:
Thanks for the tip.  

We just flattened all our submodules except OpenSTA (its a special situation).

oharboe, 2021-01-14:
Nice!

"flattened", i.e. no more git submodule then... git submodule is a disaster. git subtree doesn't incur the cognitive load for everyone.

closed 2021-04-30 by jjcherry56