The-OpenROAD-Project/OpenROAD #550: Gui tool fails to build on Ubuntu with build_openroad.sh -local

aolofsson, 2020-12-29:
Could be use user error...but the gui fails to compile on Ubuntu 20.04? 

During buid can't find "tcl.h".

Noticed that most openroad Cmake files now call out -I/usr/include/tcl, but not the gui tool.

Seems to trace back to this file?

https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/src/gui/CMakeLists.txt

(Hack to get through cmake, but this is probably not the right approach:-))

target_include_directories(gui
  PUBLIC
    ${GUI_HOME}/include
    /usr/include/tcl                                               <--ADDED THIS LINE TO MAKE IT WORK
    ${OPENDB_HOME}/include
  PRIVATE
    ${OPENROAD_HOME}/include
    ${Boost_INCLUDE_DIRS}
)

maliberty, 2020-12-29:
We build nightly on ubuntu 20 and I am able to do so as well.  What version of ubuntu are you using?  Are you using the head of the master branch?

It is better to use ${TCL_INCLUDE_PATH}.

aolofsson, 2020-12-29:
I am running 20.04. Do you put the TCL_INCLUDE_PATH go in the env setup? (.bashrc)

maliberty, 2020-12-29:
We build nightly on 20.04 too.  See https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/jenkins/docker/Dockerfile.ubuntu20.gcc8

${TCL_INCLUDE_PATH} would be better in the gui/CMakeLists.txt than /usr/include/tcl.

What version of cmake are you using?

aolofsson, 2021-01-02:
Thanks for the pointer to the dockerfile! All set and running through passing tests. :thumbsup:

closed 2021-01-02 by aolofsson