The-OpenROAD-Project/OpenROAD #551: Tautology Warning in LEF Writer

QuantamHD, 2020-12-29:
When compiling OpenDB under clang I noticed that these checks always evaluates to false, and wondered if they need to be changed. It appears they are doing a range check, in which case one of the comparison operators needs to be changed depending on if the range is supposed to be closed (2,4) or (-inf, 2) U (4, inf).

I can submit a PR if the case is clear, but I thought it should be brought to the maintainer's attention.

```
src/OpenDB/src/lef/lef/lefwWriter.cpp:1133:25: warning: overlapping comparisons always evaluate to false [-Wtautological-overlap-compare]
        if (viaCuts < 2 && viaCuts > 4)
            ~~~~~~~~~~~~^~~~~~~~~~~~~~
src/OpenDB/src/lef/lef/lefwWriter.cpp:1125:25: warning: overlapping comparisons always evaluate to false [-Wtautological-overlap-compare]
        if (viaCuts < 2 && viaCuts > 4)
            ~~~~~~~~~~~~^~~~~~~~~~~~~~
```

maliberty, 2020-12-29:
Thanks for the report.

This should be || not &&.  This fine code comes straight from Si2 but we'll fix our copy.

QuantamHD, 2020-12-29:
Is there a repo where the changes can be submitted upstream?

maliberty, 2020-12-29:
Not a public one.  You could report it to https://si2.org  (who would in turn report it to Cadence).

closed 2021-02-18 by QuantamHD