The-OpenROAD-Project/OpenROAD #552: global_placement fails without -disable_routability_driven option

antonblanchard, 2020-12-29:
I removed the `-disable_routability_driven` option from the tests:

```diff --git a/test/flow.tcl b/test/flow.tcl
index 86ea4639..29a7bd77 100644
--- a/test/flow.tcl
+++ b/test/flow.tcl
@@ -20,7 +20,7 @@ place_pins -random -hor_layers $io_placer_hor_layer -ver_layers $io_placer_ver_l
 # Macro Placement
 if { [have_macros] } {
   # tdms_place (but replace isn't timing driven)
-  global_placement -disable_routability_driven -density $global_place_density
+  global_placement -density $global_place_density
 
   macro_placement -global_config $ip_global_cfg
 }
@@ -35,7 +35,7 @@ pdngen -verbose $pdn_cfg
 
 ################################################################
 # Global placement
-global_placement -disable_routability_driven \
+global_placement \
   -density $global_place_density \
   -init_density_penalty $global_place_density_penalty \
   -pad_left $global_place_pad -pad_right $global_place_pad
```

And they fail during routing. eg:

```
../build/src/openroad gcd_sky130.tcl

...

post process guides ...
GCELLGRID X 0 DO 41 STEP 14400 ;
GCELLGRID Y 0 DO 41 STEP 14400 ;
  complete FR_MASTERSLICE
  complete FR_VIA
  complete li1
  complete mcon
  complete met1
  complete via
  complete met2
  complete via2
  complete met3
  complete via3
  complete met4
  complete via4
  complete met5
Error: pin _370_/Y not covered by guide
Error: pin _371_/D not covered by guide
Error: pin _370_/Y not covered by guide
Error: pin _371_/D not covered by guide
Warning: _000_ 2 pin not visited, fall back to feedrough mode
Error: _000_ 2 pin not visited #guides = 5
Error: critical error guide not connected, exit now 1!
```

maliberty, 2021-01-05:
@eder-matheus this looks like a bad guide from FR.

eder-matheus, 2021-01-06:
@antonblanchard Fix for the issue will be available in the next update of the public repo.

antonblanchard, 2021-03-06:
This looks to be fixed, closing

closed 2021-03-06 by antonblanchard