The-OpenROAD-Project/OpenROAD #554: pdngen requires viarule layer order that is not guaranteed

juspertor, 2021-01-02:
**pdngen** requires a certain layer order in the TLEF like:
```
 VIARULE ruleVia23 GENERATE
    LAYER MET2 ;
        ENCLOSURE 0.2 0.2 ;

    LAYER MET3 ;
        ENCLOSURE 0.2 0.2 ;

    LAYER VIA2 ;
        RECT -0.25 -0.25 0.25 0.25 ;
        SPACING 1.2  BY 1.2  ;
        RESISTANCE 1.2 ;
END ruleVia23
```

It fails for example if the upper layer is listed first:
```
VIARULE ruleVia23 GENERATE
    LAYER MET3 ;
        ENCLOSURE 0.2 0.2 ;

    LAYER MET2 ;
        ENCLOSURE 0.2 0.2 ;

    LAYER VIA2 ;
        RECT -0.25 -0.25 0.25 0.25 ;
        SPACING 1.2  BY 1.2  ;
        RESISTANCE 1.2 ;
END ruleVia23
```
The required order is common, but not guaranteed. So it needs to be checked when parsing the viarule which one is lower and upper layer. This malfunction can be fixed by replacing *proc init_via_tech* in the file *PdnGen.tcl* with the version below:

```
proc init_via_tech {} {
  variable tech
  variable def_via_tech
  
  set def_via_tech {}
  foreach via_rule [$tech getViaGenerateRules] {
    set level0 [[[$via_rule getViaLayerRule 0] getLayer] getNumber]
    set level1 [[[$via_rule getViaLayerRule 1] getLayer] getNumber]
    set level2 [[[$via_rule getViaLayerRule 2] getLayer] getNumber]
    if {$level0<$level1 && $level0<$level2} {
       set lower [$via_rule getViaLayerRule 0]
       if {$level1<$level2} {
          set upper [$via_rule getViaLayerRule 2]
          set cut   [$via_rule getViaLayerRule 1]
       } else {
          set upper [$via_rule getViaLayerRule 1]
          set cut   [$via_rule getViaLayerRule 2]
       }
    } elseif {$level1<$level0 && $level1<$level2} {
       set lower [$via_rule getViaLayerRule 1]
       if {$level0<$level2} {
          set upper [$via_rule getViaLayerRule 2]
          set cut   [$via_rule getViaLayerRule 0]
       } else {
          set upper [$via_rule getViaLayerRule 0]
          set cut   [$via_rule getViaLayerRule 2]
       }
    } elseif {$level2<$level0 && $level2<$level1} {
       set lower [$via_rule getViaLayerRule 2]
       if {$level0<$level1} {
          set upper [$via_rule getViaLayerRule 1]
          set cut   [$via_rule getViaLayerRule 0]
       } else {
          set upper [$via_rule getViaLayerRule 0]
          set cut   [$via_rule getViaLayerRule 1]
       }
    }
    
    dict set def_via_tech [$via_rule getName] [list \
      lower [list layer [[$lower getLayer] getName] enclosure [$lower getEnclosure]] \
      upper [list layer [[$upper getLayer] getName] enclosure [$upper getEnclosure]] \
      cut   [list layer [[$cut getLayer] getName] spacing [$cut getSpacing] size [list [[$cut getRect] dx] [[$cut getRect] dy]]] \
    ]
    
  }
}
```

Colin-Holehouse, 2021-01-06:
Thanks for the bug report. A fix has been added to the pdngen code, so this issue should now be resolved. Please let me know if there are any futher issues.

maliberty, 2021-01-06:
@tspyrou to close once this is pushed to the public repo

closed 2021-04-20 by maliberty