The-OpenROAD-Project/OpenROAD #561: FastRoute routing over obstructions

ax3ghazy, 2021-01-11:
I simplified the test case as far as I could.

[fastroute_through_obs.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/5797936/fastroute_through_obs.tar.gz)

To reproduce, run the script included and view the guides produced for the "la_data_in[9]" net. This misguides TritonRoute causing shorts through obstructed macros. 

[image]

juspertor, 2021-01-13:
I can confirm this issue. Obstruction macros are not respected and also pdn-rails are ignored. I can supply further test cases if needed. 

[image]

red marked shapes were added by TritonRoute and causing a short

[image]

eder-matheus, 2021-01-21:
@ax3ghazy FastRoute creates _met4_ guides over the _met4_ macro obstructions because there are free tracks between these obstructions. If you zoom the macro obs, you will be able to see it.

I'd suggest you use the "-macro_extension _n_" option. With this, the macro obstacles are extended in _n_ gcells. In my tests, using "-macro_extension 1" was enough to avoid routing over the macro obs.

closed 2021-07-08 by eder-matheus