The-OpenROAD-Project/OpenROAD #562: Tapping distance not being strictly applied

ax3ghazy, 2021-01-11:
The attached test case specifies a 14um tapping distance, which is not being strictly applied

[tapping_distance_issue.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/5797974/tapping_distance_issue.tar.gz)

To reproduce, run the attached script and check the distances between taps/end-caps, which exceed the specified 14um.

Moreover, it's not completely clear to me what type of tap spacing rule is being targeted by the tool, but I believe that
it should eventually implement (or support) the worst case scenario (end-of-tap to end-of-cell spacing).

[image]

eder-matheus, 2021-01-21:
The fix for this issue is already available at the master branch.

ax3ghazy, 2021-02-17:
The fix improves things, but the same test case still fails (distance is still being above the specified 14um).

[image]

closed 2021-04-20 by maliberty