The-OpenROAD-Project/OpenROAD #566: TritonRoute SEGV when loading DEF

antonblanchard, 2021-01-24:
To reproduce:

```
touch empty.lef empty.def empty.guide
cat << EOF > test.param
lef:empty.lef
def:empty.def
guide:empty.guide
output:out.def
outputTA:outTA.def
outputguide:out.guide
outputDRC:out.drc
outputMaze:out.log
threads:8
verbose:1
drouteEndIterNum:64
EOF
echo tr::detailed_route_cmd test.param > test.tcl
openroad test.tcl
```

I bisected it back to commit 6d7b5d99ff21 ("fix comp placement issue")

antonblanchard, 2021-01-24:
```
(gdb) backtrace 
#0  0x0000000000743050 in odb::dbChip::getBlock() ()
#1  0x0000000000b32a26 in fr::io::Parser::readDb(odb::dbDatabase*) ()
#2  0x0000000000b32f60 in fr::io::Parser::readLefDb(odb::dbDatabase*) ()
#3  0x0000000000a40575 in triton_route::TritonRoute::init() ()
#4  0x000000000046a3b3 in triton_route::TritonRoute::main() ()
#5  0x0000000000d8ac0d in _wrap_detailed_route_cmd ()
#6  0x00007ffff7e34042 in TclNRRunCallbacks () from /lib64/libtcl8.6.so
#7  0x00007ffff7e39da7 in ?? () from /lib64/libtcl8.6.so
#8  0x00007ffff7e3a5d7 in Tcl_EvalEx () from /lib64/libtcl8.6.so
#9  0x00007ffff7e3a5fa in Tcl_Eval () from /lib64/libtcl8.6.so
#10 0x000000000058971c in sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) ()
#11 0x0000000000471bb9 in tclAppInit(int, char**, char const*, Tcl_Interp*) ()
#12 0x00007ffff7ee9ee6 in Tcl_MainEx () from /lib64/libtcl8.6.so
#13 0x0000000000464c2e in main ()
```

antonblanchard, 2021-01-24:
@osamahammad21 I presume we require the lef and def to be read in via `read_def/read_lef` now? If so, could we warn or error if either appears in the param file?

maliberty, 2021-01-25:
Yes we are in transition to using the db for all lef/def.  For now you have to provide both.  We'll add a check/message.

antonblanchard, 2022-10-03:
Now the detailed router is properly integrated into OpenROAD, this is no longer relevant. Closing.

closed 2022-10-03 by antonblanchard