The-OpenROAD-Project/OpenROAD #567: OpenDB produces LEF output that openphysyn cannot understand

antonblanchard, 2021-01-25:
I tried updating the version of openroad in openlane, and it fails in openphysyn:

```
Reading DEF file: /openLANE_flow/designs/multiply_4/runs/25-01_02-40/tmp/placement/11-replace.def
Notice 0: Design: multiply_4
Notice 0: error: pin with multiple ports is not supported
ERROR (DEFPARS-6010): An error has been reported in callback. Error in file /openLANE_flow/designs/multiply_4/runs/25-01_02-40/tmp/placement/11-replace.def at line 61239, on token ;.
Update the def file before parsing the file again.
Notice 0: DEF parser returns an error![ERROR]: during executing: "Psn /openLANE_flow/scripts/openPhySyn.tcl |& tee >&@stdout /openLANE_flow/designs/multiply_4/runs/25-01_02-40/logs/placement/12-openphysyn.log"
```

The relevant part of the LEF file:

```
    - vccd1 + NET vccd1 + SPECIAL + DIRECTION INOUT + USE POWER
      + PORT
        + LAYER met4 ( -800 -488480 ) ( 800 488480 )
        + FIXED ( 943440 499120 ) N
      + PORT
        + LAYER met4 ( -800 -488480 ) ( 800 488480 )
        + FIXED ( 789840 499120 ) N
```

The older version of openroad split these up with extra* suffixes:

```
- vccd1 + NET vccd1 + DIRECTION INOUT + USE POWER
  + LAYER met4 ( -800 -329360 ) ( 800 329360 )
  + FIXED ( 636240 340000 ) N + SPECIAL ;
- vccd1.extra1 + NET vccd1 + DIRECTION INOUT + USE POWER
  + LAYER met4 ( -800 -329360 ) ( 800 329360 )
  + FIXED ( 482640 340000 ) N + SPECIAL ;

```

I'm trying to understand what 40722f087fb1 ("Revert "Merge branch 'master' into openroad"") is doing. It claims to be a revert, but it wrapped the code that emits the extra* suffixes with:

```
+  if(cnt == 0 || _version <= defout::DEF_5_6){
```

Which I can't find previously in the history nor in OpenDB (so is it really a revert?) . Was this intended?

maliberty, 2021-01-25:
I would guess openphysyn is using on old version of opendb.  The change is intended and is legal DEF.  I suggest you report the problem to openphysyn.

closed 2021-01-25 by maliberty

antonblanchard, 2021-01-25:
I pulled OpenDB from https://github.com/The-OpenROAD-Project/OpenDB and I see the same issue:

```
# cat x.tcl 
set db [odb::dbDatabase_create]
odb::read_lef $db [file join "merged_unpadded.lef" ]
odb::read_def $db [file join "11-replace.def"]

# opendbtcl x.tcl
Notice 0: Reading LEF file:  merged_unpadded.lef
Notice 0:     Created 13 technology layers
Notice 0:     Created 25 technology vias
Notice 0:     Created 438 library cells
Notice 0: Finished LEF file:  merged_unpadded.lef
Notice 0: 
Reading DEF file: 11-replace.def
Notice 0: Design: multiply_4
Notice 0: error: pin with multiple ports is not supported
ERROR (DEFPARS-6010): An error has been reported in callback. Error in file 11-replace.def at line 61239, on token ;.
Update the def file before parsing the file again.
Notice 0: DEF parser returns an error!
```

Is there another OpenDB tree I should be looking at?

antonblanchard, 2021-01-25:
I thought that tree might now be dead, and `openroad/src/OpenDB` is the canonical location. That doesn't build however:

```
OpenROAD-test/src/OpenDB/src/defout/defout_impl.cpp:44:10: fatal error: utility/Logger.h: No such file or directory
   44 | #include "utility/Logger.h"
      |          ^~~~~~~~~~~~~~~~~~
```

ax3ghazy, 2021-01-25:
@antonblanchard: We needed to install `qt5-qtbase-devel` (on CentOS for example) to get around that one...

maliberty, 2021-01-25:
The standalone repo is not being updated.  That file is present in the repo: https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/src/utility/include/utility/Logger.h

The logger has noting to do with Qt.

ax3ghazy, 2021-01-25:
@maliberty: Yea, somehow I confused this with another issue I was getting; I used to get
```
OpenROAD/src/replace/src/replace.cpp:40:10: fatal error: utility/Logger.h: No such file or directory
```
until I installed the Qt devel package. CMake was somehow back then producing Makefiles with compile commands that don't have the required `-lutility`.

However, the issue that @antonblanchard flagged is a different one. I can see the same error trying to build OpenDB independently inside the cloned OpenROAD repo following the standard instructions on its README:

```
cd src/OpenDB
mkdir build
cd build
cmake ..
make
```

This fails with the error above, which makes sense since "utility" isn't a target in its CMakeLists.txt... Not sure if I am missing something, but I needed a couple of modifications to get it to build. Here's a [draft](https://gist.github.com/ax3ghazy/8c7ac9aa5c1341f6d528c5b6934b454d) of those modifications. I also needed to build `spdlog` with `-DSPDLOG_BUILD_SHARED=ON`. Of course, those changes break the rest of the build since the target `utility` is duplicated, so I am not sure what the right solution is. Is the latest commit passing your CI opendbpy regression tests?

maliberty, 2021-01-25:
We are not building opendb as a standalone module only as a part of openroad.  opendbpy was never a production interface and is not built in our CI.  

If you just want to build the db you should still run cmake at the top level application and just run 'make opendb'.