The-OpenROAD-Project/OpenROAD #568: Infinite loop found in Resizer command

QuantamHD, 2021-02-01:
Version: 0da4562b518d88ff95e086ea1350b4e96156aa99

I've attached an archive of the [Uploading openroad_infite_loop.tar.gz…](Infinite loop reproduction). I've debugged the code with lldb, and narrowed it down to this section of code. 
```c++
  //Resizer.cc:552
  int max_length = metersToDbu(max_wire_length);
  for (int i = level_drvr_vertices_.size() - 1; i >= 0; i--) {
    logger_->info(RSZ, 553, "Interation {}.", i);
    if (i == 959) {
      logger_->info(RSZ, 630, "Interation {}.", i);
    }
    Vertex *drvr = level_drvr_vertices_[i];
    Pin *drvr_pin = drvr->pin();
    Net *net = network_->net(drvr_pin);
    if (net &&
        !sta_->isClock(drvr_pin)
        // Exclude tie hi/low cells.
        && !isFuncOneZero(drvr_pin) && !isSpecial(net)) {
      repairNet(net, drvr, true, true, true, max_length, true, repair_count,
                slew_violations, cap_violations, fanout_violations,
                length_violations);
    }
  }
```

On the 959th iteration of this loop it will get stuck in an infinite loop forever. It gets stuck specifically in this routine in the while loop.

```c++
void
Network::visitConnectedPins(const Net *net,
			    PinVisitor &visitor,
			    ConstNetSet &visited_nets) const
{
  if (!visited_nets.hasKey(net)) {
    visited_nets.insert(net);
    // Search up from net terminals.
    NetTermIterator *term_iter = termIterator(net);
    while (term_iter->hasNext()) {
      Term *term = term_iter->next();
      Pin *above_pin = pin(term);
      if (above_pin) {
	Net *above_net = this->net(above_pin);
	if (above_net)
	  visitConnectedPins(above_net, visitor, visited_nets);
	else
	  visitor(above_pin);
      }
    }

```

Where in the NetTermIterator will return a cycle. Each call to next will return a pointer to the other 0x01 -> 0x02 -> 0x01 -> 0x02 -> ... forever.

The stack trace to the infinite loop
[image]

QuantamHD, 2021-02-01:
After further investigation it appears that the net in question is named "net285"

jjcherry56, 2021-02-02:
I got that. The issue is upstream of Network::visitConnectedPins. It is just the victim.

jjcherry56, 2021-02-02:
7dec39f6 dbBTerm::connect missing _next_bterm init issue#568

closed 2021-02-02 by jjcherry56

QuantamHD, 2021-02-02:
Is the dbBTerm::connect issue tracked elsewhere?

jjcherry56, 2021-02-02:
why does it have to be tracked? I showed you the commit that fixed it.

QuantamHD, 2021-02-02:
I apologize, it's just that I wasn't sure which repository 7dec39f6 was in as it didn't automatically link to any commit in the current repo. Is it in the openroad branch?

I was just trying to take a look at the diff and I wasn't sure where it was located or if that was indeed the fix.

Thanks for taking a look at it.

jjcherry56, 2021-02-02:
It gets pushed to the private repo and from there to the public one that you see, usually weekly.