The-OpenROAD-Project/OpenROAD #570: read_elf error when  Floorplanning

troysung, 2021-02-07:
[INFO]: Synthesis was successful
[INFO]: Running Floorplanning...
[INFO]: Running Initial Floorplanning...
[INFO]: current step index: 3
[INFO]: /openLANE_flow/scripts
OpenROAD 0.9.0 1415572a73
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
Warning: /root/ic/skywater-pdk/sky130A/libs.ref/sky130_fd_sc_hd/lib/sky130_fd_sc_hd__tt_025C_1v80.lib line 31, default_operating_condition tt_025C_1v80 not found.
Notice 0: Reading LEF file:  /openLANE_flow/designs/inverter/runs/07-02_03-56/tmp/merged_unpadded.lef
Notice 0:     Created 11 technology layers
Notice 0:     Created 25 technology vias
Notice 0:     Created 437 library cells
Notice 0: Finished LEF file:  /openLANE_flow/designs/inverter/runs/07-02_03-56/tmp/merged_unpadded.lef
[ERROR]: during executing: "openroad -exit /openLANE_flow/scripts/openroad/or_floorplan.tcl |& tee >&@stdout /openLANE_flow/designs/inverter/runs/07-02_03-56/logs/floorplan/3-verilog2def.openroad.log"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child killed: segmentation violation

[ERROR]: Please check openroad  log file
[ERROR]: Dumping to /openLANE_flow/designs/inverter/runs/07-02_03-56/error.log

troysung, 2021-02-07:
maybe it is because of "read_lef" itself, and I cannot find where "read_lef" is

maliberty, 2021-02-07:
Please provide a test case that reproduces the problem.

troysung, 2021-02-07:
the sky130 pdk was generated by others,as the pdk source files are too large to download.
docker image is v0.2.
exec "./flow.tcl -design picorv32a" or almost every project in the "designs" directory my cause this problem.
could it be the pdk problem?

maliberty, 2021-02-24:
read_lef is in src/OpenRoad.tcl

maliberty, 2021-06-25:
We run with the sky130 lefs for hd & hs.  Please provide a lef file that shows the issue or let's close this.

closed 2021-06-28 by maliberty