The-OpenROAD-Project/OpenROAD #572: [1.0rc2] ld: error: undefined symbol: sta::elapsedRunTime()

yurivict, 2021-02-13:
```
[451/664] : && /usr/local/bin/cmake -E rm -f src/OpenSTA/sta_swig.a && /usr/local/bin/ar qc src/OpenSTA/sta_swig.a  src/OpenSTA/CMakeFiles/sta_swig.dir/CMakeFiles/sta_swig.dir/StaAppTCL_wrap.cxx.o && /usr/local/bin/ranlib src/OpenSTA/sta_swig.a && :
[452/664] : && /usr/bin/c++ -O2 -pipe -fno-omit-frame-pointer -fstack-protector-strong -isystem /usr/local/include -fno-strict-aliasing -fno-omit-frame-pointer  -I/usr/local/include/eigen3 -isystem /usr/local/include -O2 -pipe -fno-omit-frame-pointer -fstack-protector-strong -isystem /usr/local/include -fno-strict-aliasing -fno-omit-frame-pointer  -I/usr/local/include/eigen3 -isystem /usr/local/include -fstack-protector-strong -L/usr/local/lib src/OpenSTA/CMakeFiles/sta.dir/app/Main.cc.o -o /usr/ports/cad/openroad/work/OpenROAD-1.0rc2/src/OpenSTA/app/sta  src/OpenSTA/sta_swig.a  /usr/ports/cad/openroad/work/OpenROAD-1.0rc2/src/OpenSTA/app/libOpenSTA.a  /usr/local/lib/libtcl86.so  -pthread  /usr/lib/libz.so  /usr/local/lib/libcudd.so && :
FAILED: /usr/ports/cad/openroad/work/OpenROAD-1.0rc2/src/OpenSTA/app/sta 
: && /usr/bin/c++ -O2 -pipe -fno-omit-frame-pointer -fstack-protector-strong -isystem /usr/local/include -fno-strict-aliasing -fno-omit-frame-pointer  -I/usr/local/include/eigen3 -isystem /usr/local/include -O2 -pipe -fno-omit-frame-pointer -fstack-protector-strong -isystem /usr/local/include -fno-strict-aliasing -fno-omit-frame-pointer  -I/usr/local/include/eigen3 -isystem /usr/local/include -fstack-protector-strong -L/usr/local/lib src/OpenSTA/CMakeFiles/sta.dir/app/Main.cc.o -o /usr/ports/cad/openroad/work/OpenROAD-1.0rc2/src/OpenSTA/app/sta  src/OpenSTA/sta_swig.a  /usr/ports/cad/openroad/work/OpenROAD-1.0rc2/src/OpenSTA/app/libOpenSTA.a  /usr/local/lib/libtcl86.so  -pthread  /usr/lib/libz.so  /usr/local/lib/libcudd.so && :
ld: error: undefined symbol: sta::elapsedRunTime()
>>> referenced by StaAppTCL_wrap.cxx
>>>               StaAppTCL_wrap.cxx.o:(elapsed_run_time()) in archive src/OpenSTA/sta_swig.a
>>> referenced by StaAppTCL_wrap.cxx
>>>               StaAppTCL_wrap.cxx.o:(_wrap_elapsed_run_time(void*, Tcl_Interp*, int, Tcl_Obj* const*)) in archive src/OpenSTA/sta_swig.a
>>> referenced by Stats.cc
>>>               Stats.cc.o:(sta::Stats::Stats(sta::Debug*)) in archive /usr/ports/cad/openroad/work/OpenROAD-1.0rc2/src/OpenSTA/app/libOpenSTA.a
>>> referenced by Stats.cc
>>>               Stats.cc.o:(sta::Stats::report(char const*)) in archive /usr/ports/cad/openroad/work/OpenROAD-1.0rc2/src/OpenSTA/app/libOpenSTA.a

ld: error: undefined symbol: sta::userRunTime()
>>> referenced by StaAppTCL_wrap.cxx
>>>               StaAppTCL_wrap.cxx.o:(user_run_time()) in archive src/OpenSTA/sta_swig.a
>>> referenced by StaAppTCL_wrap.cxx
>>>               StaAppTCL_wrap.cxx.o:(cputime()) in archive src/OpenSTA/sta_swig.a
>>> referenced by StaAppTCL_wrap.cxx
>>>               StaAppTCL_wrap.cxx.o:(_wrap_user_run_time(void*, Tcl_Interp*, int, Tcl_Obj* const*)) in archive src/OpenSTA/sta_swig.a
>>> referenced by StaAppTCL_wrap.cxx
>>>               StaAppTCL_wrap.cxx.o:(_wrap_cputime(void*, Tcl_Interp*, int, Tcl_Obj* const*)) in archive src/OpenSTA/sta_swig.a
>>> referenced by Stats.cc
>>>               Stats.cc.o:(sta::Stats::Stats(sta::Debug*)) in archive /usr/ports/cad/openroad/work/OpenROAD-1.0rc2/src/OpenSTA/app/libOpenSTA.a
>>> referenced by Stats.cc
>>>               Stats.cc.o:(sta::Stats::report(char const*)) in archive /usr/ports/cad/openroad/work/OpenROAD-1.0rc2/src/OpenSTA/app/libOpenSTA.a

ld: error: undefined symbol: sta::memoryUsage()
>>> referenced by StaAppTCL_wrap.cxx
>>>               StaAppTCL_wrap.cxx.o:(memory_usage()) in archive src/OpenSTA/sta_swig.a
>>> referenced by StaAppTCL_wrap.cxx
>>>               StaAppTCL_wrap.cxx.o:(_wrap_memory_usage(void*, Tcl_Interp*, int, Tcl_Obj* const*)) in archive src/OpenSTA/sta_swig.a
>>> referenced by Stats.cc
>>>               Stats.cc.o:(sta::Stats::Stats(sta::Debug*)) in archive /usr/ports/cad/openroad/work/OpenROAD-1.0rc2/src/OpenSTA/app/libOpenSTA.a
>>> referenced by Stats.cc
>>>               Stats.cc.o:(sta::Stats::report(char const*)) in archive /usr/ports/cad/openroad/work/OpenROAD-1.0rc2/src/OpenSTA/app/libOpenSTA.a

ld: error: undefined symbol: sta::processorCount()
>>> referenced by StaAppTCL_wrap.cxx
>>>               StaAppTCL_wrap.cxx.o:(processor_count()) in archive src/OpenSTA/sta_swig.a
>>> referenced by StaAppTCL_wrap.cxx
>>>               StaAppTCL_wrap.cxx.o:(_wrap_processor_count(void*, Tcl_Interp*, int, Tcl_Obj* const*)) in archive src/OpenSTA/sta_swig.a
>>> referenced by StaMain.cc
>>>               StaMain.cc.o:(sta::parseThreadsArg(int&, char**)) in archive /usr/ports/cad/openroad/work/OpenROAD-1.0rc2/src/OpenSTA/app/libOpenSTA.a

ld: error: undefined symbol: sta::initElapsedTime()
>>> referenced by Sta.cc
>>>               Sta.cc.o:(sta::initSta()) in archive /usr/ports/cad/openroad/work/OpenROAD-1.0rc2/src/OpenSTA/app/libOpenSTA.a

ld: error: undefined symbol: sta::systemRunTime()
```

FreeBSD 12.2
clang-10

maliberty, 2021-02-13:
does this happen in a clean build?  sometimes the swig wrappers don't update when they should.

yurivict, 2021-02-13:
Yes, this is in a clean build.

maliberty, 2021-02-14:
@jjcherry56 any ideas?

jjcherry56, 2021-02-15:
We don't/can't support free bsd because we don't have anything running it and aws does not even have an image for it. I may have fixed this, but honestly I am tired of seeing free bsd porting issues. How do we discontinue the free bsd port? Why does it exist?

closed 2021-02-15 by jjcherry56

yurivict, 2021-02-15:
Why don't you discontinue the project altogether?

maliberty, 2021-02-15:
Yikes.  I see its defined in OpenSTA/util/Machine.cc.  Do you see the symbol there in your build?  You can check the .o's with nm.