The-OpenROAD-Project/OpenROAD #573: report_checks > filename, not working properly.

agorararmard, 2021-02-15:
Hi all, using `report_checks > filename` used to work inside openroad, but for some reason, it's not working anymore. This might've happened sometime during the last 2 months. Now the following code segment is producing the reports inside the log file instead of the designated report files:

```tcl
report_checks -fields {capacitance slew input_pins nets fanout} -unique -slack_max -0.0 -group_count 100 > $::env(cts_report_file_tag).timing.rpt
report_checks -fields {capacitance slew input_pins nets fanout} -path_delay min_max > $::env(cts_report_file_tag).min_max.rpt
report_checks -fields {capacitance slew input_pins nets fanout} -group_count 100  -slack_max -0.01 > $::env(cts_report_file_tag).rpt

report_wns > $::env(cts_report_file_tag)_wns.rpt
report_tns > $::env(cts_report_file_tag)_tns.rpt
report_clock_skew > $::env(cts_report_file_tag)_clock_skew.rpt
```

Is there something that I'm doing totally wrong?

tspyrou, 2021-02-15:
This is a side effect of moving to the new unified logger spdlog. We will need to enhance spdlog to  support tcl redirects before we can support his again. For now all the information goes to the log file.

jjcherry56, 2021-02-15:
opensta specific log command support (ie, redirection) was abandoned when spdlog support was added to openroad.

closed 2021-04-20 by maliberty