The-OpenROAD-Project/OpenROAD #574: Tritonroute FlexDRWorker::mazeIterInit_sortRerouteNets integer overflow

antonblanchard, 2021-02-16:
I ran the LLVM undefined behaviour sanitizer on OpenROAD and it found an integer overflow in `FlexDRWorker::mazeIterInit_sortRerouteNets`:

```
bool FlexDRWorker::mazeIterInit_sortRerouteNets(int mazeIter, vector<drNet*> &rerouteNets) {
  auto rerouteNetsComp1 = [](drNet* const &a, drNet* const &b) {
                         frBox boxA, boxB;
                         a->getPinBox(boxA);
                         b->getPinBox(boxB);
                         auto areaA = (boxA.right() - boxA.left()) * (boxA.top() - boxA.bottom());
                         auto areaB = (boxB.right() - boxB.left()) * (boxB.top() - boxB.bottom());
```

Since `left/right/top/bottom` are signed integers, the multiplication will be truncated to a signed integer and `area*` will also be signed integers. Apart from being undefined, it seems likely that `area*` will wrap negative, and the signed comparison right after will be wrong.

A potential fix follows. We need to cast one of the arguments to a larger type, as well as specifying a larger type for `area*`. I can submit a PR along these lines if wanted:

```
@@ -1513,8 +1513,8 @@ bool FlexDRWorker::mazeIterInit_sortRerouteNets(int mazeIter, vector<drNet*> &re
                          frBox boxA, boxB;
                          a->getPinBox(boxA);
                          b->getPinBox(boxB);
-                         auto areaA = (boxA.right() - boxA.left()) * (boxA.top() - boxA.bottom());
-                         auto areaB = (boxB.right() - boxB.left()) * (boxB.top() - boxB.bottom());
+                         unsigned long long areaA = (unsigned long long)(boxA.right() - boxA.left()) * (boxA.top() - boxA.bottom());
+                         unsigned long long areaB = (unsigned long long)(boxB.right() - boxB.left()) * (boxB.top() - boxB.bottom());
```

maliberty, 2021-02-16:
Thanks for the testing!  I think the fix is simple enough I'll just do it myself.

closed 2021-02-17 by maliberty

antonblanchard, 2021-02-17:
Thanks @maliberty!