The-OpenROAD-Project/OpenROAD #575: FastRoute coeffADJ() undefined behaviour

antonblanchard, 2021-02-17:
The LLVM undefined behaviour sanitizer is complaining about integer overflows in `gen_brk_RSMT()`

```
../src/FastRoute/src/fastroute/src/RSMT.cpp:428:34: runtime error: inf is outside the range of representable values of type 'int'
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior ../src/FastRoute/src/fastroute/src/RSMT.cpp:428:34 in
../src/FastRoute/src/fastroute/src/RSMT.cpp:428:25: runtime error: signed integer overflow: 282 * -2147483648 cannot be represented in type 'int'
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior ../src/FastRoute/src/fastroute/src/RSMT.cpp:428:25 in
```

The issue appears to be in `coeffADJ()` where a floating point divide by zero `(Husage * Vcap)`  will result in `-inf`, `inf` or `nan` depending on the numerator.

I'm a bit lost as to what this function is meant to do - it caps the minimum but has no cap on the maximum (so `inf` is a valid adjustment factor). If it is trying to create a scaling factor around 1.0, perhaps it should limit the result to 0.8 - 1.2. It needs to also handle the zero divided by zero = `nan` case.

```
diff --git a/src/FastRoute/src/fastroute/src/RSMT.cpp b/src/FastRoute/src/fastroute/src/RSMT.cpp
index 82c00ac0..c045815a 100644
--- a/src/FastRoute/src/fastroute/src/RSMT.cpp
+++ b/src/FastRoute/src/fastroute/src/RSMT.cpp
@@ -791,7 +791,11 @@ float coeffADJ(int netID)
     coef = (Hcap * Vusage) / (Husage * Vcap);
   }
 
-  if (coef < 1.2) {
+  if (std::isnan(coef)) {
+    coef = 1.0;
+  } else if (coef < 0.8) {
+    coef = 0.8;
+  } else if (coef > 1.2) {
     coef = 1.2;
   }
 
```
I can create a PR if this looks reasonable, or adjust it as required.

maliberty, 2021-02-17:
Thanks for the help.  I am not an expert on this code but I believe the idea is to account for congestion when building the Steiner tree using flute.  Since flute is not congestion aware they stretch the inputs according to the H vs V congestion to make it prefer the less congested direction.  'coef' is the stretching factor.

I suspect this is rarely a problem is practice but I don't know for sure (Vcap should never be zero and Husage being zero would be rare).   

For a change like this I think you need to run a bunch of designs from OpenROAD-flow-scripts and make sure there is no degradation.  I've assigned this to Eder for testing unless you want to take that on.

jjcherry56, 2021-07-06:
I don't think it makes any sense to divide by zero and then test for nan.
Check for the zero before dividing.

closed 2021-07-07 by eder-matheus