The-OpenROAD-Project/OpenROAD #578: write_verilog flips the indices of big-endian ports

Manarabdelaty, 2021-02-19:
write_verilog always assumes that ports are little-endian. In the attached test case, the original netlist has one big-endian port `input [0:1] x;` which is written as `input [1:0] x;` in the output netlist.

[spm_test_case.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/6009866/spm_test_case.zip)

mooredan, 2021-07-29:
Sharing the contents of the demonstration script for discussion:

```
read_verilog spm.v
read_lef merged.lef
read_def spm.def

write_verilog spm_out.v 
```

If this were modified to:

```
read_verilog spm.v
write_verilog spm_out.v
```

no output is produced.  Therefore, reading a verilog file does not sufficiently populate a database that could be written out, other processing is needed. 

Reading this particular DEF file holds an alternate (maybe the same) representation of the spm design.  Reading the DEF does sufficiently populate the database.  If the command file where changed to:

```
read_lef merged.lef
read_def spm.def
write_verilog spm_out.v
```

Will produce the exact same verilog file as when the read_verilog command was present.

So it's probably safe to say that none of the original read verilog is retained, or considered when writing out the verilog. That is, the read_verilog command is moot.

I'm guessing that during the DEF read that the netlist is constructed.  During the write, big endian is assumed and that's why the output is as it is.

Unless the process is _supposed to_ pay attention to the verilog_read, then one way to solve this would be a write_verilog command option that indicates little endian for ports, or for specific ports.   It would be up to the user to know which ones need to be little endian and specify it.  This seems quite awkaward, but is a solution that may work.

Based upon my short engagement with this project, I suspect that this isn't something that the architects would embrace.  You're probably much better off just post-processing the resulting verilog (or adopting big endian everywhere).

jjcherry56, 2021-07-29:
I don't have to guess.
DEF and OpenDB have no notion of busses so the bus bits are stored individually.
write_verilog has to reconstruct the bus.
The correct solution to annotate the "first" port bit in the DB so it can be written correctly.

maliberty, 2021-08-10:
@jjcherry56 could you just store this in a dbBoolProperty?

closed 2021-08-13 by jjcherry56