The-OpenROAD-Project/OpenROAD #586: error in "history.tcl" when compiled with tcl_readline

yichungchen84, 2021-02-27:
when compiled with tcl_readline

openroad xxx.tcl (I ran examples from the test folder)
it ran well, but it shows "application-specific initialization failed: error while autoloading "history": Error: history.tcl, 330 invoked "return" outside of a proc." at the end

I dig into it. It is because of tcl_readline. 
remove tcl_readline form the configure and no more error.

btw:
example jpeg_sky130hd.tcl terminated due to 
Design area 8883380 u^2 100% utilization.
if { [sta::worst_slack -max] < $setup_slack_limit } {
  fail "setup slack limit exceeded [format %.2f [sta::worst_slack -max]] < $setup_slack_limit"
}
fail setup slack limit exceeded -5.24 < -1.5

maliberty, 2021-03-02:
I regularly build with tcl readline without any issue so it must be related to your setup.  Can you try it with a docker build?

yichungchen84, 2021-03-05:
> I regularly build with tcl readline without any issue so it must be related to your setup. Can you try it with a docker build?

The docker version is fine. I tested it before. 
The new build is local on a Ubuntu. It seems that history.tcl has some problems only on Ubuntu/Debian. I will to to see if I can find the root cause.

maliberty, 2021-04-20:
@yichungchen84 any further update?  If not we should close this.

rovinski, 2021-05-21:
@maliberty I actually see this on my Ubuntu 20 machine as well.

maliberty, 2021-05-21:
I can't reproduce this using Ubuntu on WSL2
% lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description: Ubuntu 20.04.2 LTS
Release: 20.04
Codename: focal

@rovinski since you can would you debug this a bit?

rovinski, 2021-05-24:
@maliberty I'm not really sure how to start tracking this down. I don't know where the `history.tcl` file exists. I do know that this only shows up for me when running an interactive shell - it doesn't happen when loading a script.

Here's the stack trace(s) from when the error is printed (breaking on syscalls to write):
```
Breakpoint 1, __libc_write (fd=1, buf=0x5555580cf838, nbytes=44) at ../sysdeps/unix/sysv/linux/write.c:25
25      ../sysdeps/unix/sysv/linux/write.c: No such file or directory.
(gdb) backtrace
#0  __libc_write (fd=1, buf=0x5555580cf838, nbytes=44) at ../sysdeps/unix/sysv/linux/write.c:25
#1  0x00007ffff7f30967 in ?? () from /usr/lib/x86_64-linux-gnu/libtcl8.6.so
#2  0x00005555559ef7e6 in sta::ReportTcl::printConsole(char const*, unsigned long) ()
#3  0x00005555558d1e84 in sta::dbStaReport::printString(char const*, unsigned long) ()
#4  0x00005555559ef5d4 in encapOutputProc ()
#5  0x00007ffff7ed4140 in ?? () from /usr/lib/x86_64-linux-gnu/libtcl8.6.so
#6  0x00007ffff7ed5bf8 in ?? () from /usr/lib/x86_64-linux-gnu/libtcl8.6.so
#7  0x00007ffff7ed55ad in Tcl_WriteChars () from /usr/lib/x86_64-linux-gnu/libtcl8.6.so
#8  0x00007ffff7ef01e2 in Tcl_MainEx () from /usr/lib/x86_64-linux-gnu/libtcl8.6.so
#9  0x000055555573a8eb in main ()
(gdb) c
Continuing.
application-specific initialization failed:
Breakpoint 1, __libc_write (fd=1, buf=0x5555580cf838, nbytes=94) at ../sysdeps/unix/sysv/linux/write.c:25
25      in ../sysdeps/unix/sysv/linux/write.c
(gdb) backtrace
#0  __libc_write (fd=1, buf=0x5555580cf838, nbytes=94) at ../sysdeps/unix/sysv/linux/write.c:25
#1  0x00007ffff7f30967 in ?? () from /usr/lib/x86_64-linux-gnu/libtcl8.6.so
#2  0x00005555559ef7e6 in sta::ReportTcl::printConsole(char const*, unsigned long) ()
#3  0x00005555558d1e84 in sta::dbStaReport::printString(char const*, unsigned long) ()
#4  0x00005555559ef5d4 in encapOutputProc ()
#5  0x00007ffff7ed4140 in ?? () from /usr/lib/x86_64-linux-gnu/libtcl8.6.so
#6  0x00007ffff7ed5bf8 in ?? () from /usr/lib/x86_64-linux-gnu/libtcl8.6.so
#7  0x00007ffff7ed5695 in Tcl_WriteObj () from /usr/lib/x86_64-linux-gnu/libtcl8.6.so
#8  0x00007ffff7ef01f5 in Tcl_MainEx () from /usr/lib/x86_64-linux-gnu/libtcl8.6.so
#9  0x000055555573a8eb in main ()
(gdb) c
Continuing.
error while autoloading "history": Error: history.tcl, 330 invoked "return" outside of a proc.
Breakpoint 1, __GI___libc_write (fd=1, buf=0x555557b0fa70, nbytes=1) at ../sysdeps/unix/sysv/linux/write.c:25
25      ../sysdeps/unix/sysv/linux/write.c: No such file or directory.
(gdb) backtrace
#0  __GI___libc_write (fd=1, buf=0x555557b0fa70, nbytes=1) at ../sysdeps/unix/sysv/linux/write.c:25
#1  0x00007ffff619000d in _IO_new_file_write (f=0x7ffff62ea6a0 <_IO_2_1_stdout_>, data=0x555557b0fa70, n=1) at fileops.c:1176
#2  0x00007ffff6191ad1 in new_do_write (to_do=1, data=0x555557b0fa70 "\nomponents of this program may be licensed under more restrictive licenses which must be honored.\nd\026d\027\204", fp=0x7ffff62ea6a0 <_IO_2_1_stdout_>)
    at libioP.h:948
#3  _IO_new_do_write (to_do=1, data=0x555557b0fa70 "\nomponents of this program may be licensed under more restrictive licenses which must be honored.\nd\026d\027\204", fp=0x7ffff62ea6a0 <_IO_2_1_stdout_>) at fileops.c:426
#4  _IO_new_do_write (fp=0x7ffff62ea6a0 <_IO_2_1_stdout_>, data=0x555557b0fa70 "\nomponents of this program may be licensed under more restrictive licenses which must be honored.\nd\026d\027\204", to_do=1) at fileops.c:423
#5  0x00007ffff6190835 in _IO_new_file_xsputn (n=1, data=<optimized out>, f=<optimized out>) at libioP.h:948
#6  _IO_new_file_xsputn (f=0x7ffff62ea6a0 <_IO_2_1_stdout_>, data=<optimized out>, n=1) at fileops.c:1197
#7  0x00007ffff6184541 in __GI__IO_fwrite (buf=0x7fffffffda60, size=1, count=1, fp=0x7ffff62ea6a0 <_IO_2_1_stdout_>) at libioP.h:948
#8  0x00005555565afd03 in spdlog::sinks::ansicolor_sink<spdlog::details::console_mutex>::log(spdlog::details::log_msg const&) ()
#9  0x000055555659de63 in spdlog::logger::sink_it_(spdlog::details::log_msg const&) ()
#10 0x000055555659d96a in spdlog::logger::log_it_(spdlog::details::log_msg const&, bool, bool) ()
#11 0x00005555558d20f2 in sta::dbStaReport::printString(char const*, unsigned long) ()
#12 0x00005555559ef5d4 in encapOutputProc ()
#13 0x00007ffff7ed4140 in ?? () from /usr/lib/x86_64-linux-gnu/libtcl8.6.so
#14 0x00007ffff7ed5bf8 in ?? () from /usr/lib/x86_64-linux-gnu/libtcl8.6.so
#15 0x00007ffff7ed55ad in Tcl_WriteChars () from /usr/lib/x86_64-linux-gnu/libtcl8.6.so
#16 0x00007ffff7ef0209 in Tcl_MainEx () from /usr/lib/x86_64-linux-gnu/libtcl8.6.so
#17 0x000055555573a8eb in main ()
```
I'm guessing the `??` are probably important but I don't have debugging information in my libtcl.so

rovinski, 2021-05-24:
Note that this is similar to #102

maliberty, 2021-05-24:
@jjcherry56 any ideas?

jjcherry56, 2021-05-24:
I have seen issues with history.tcl issues in opensta over the years but frankly they are mysterious. history.tcl is one of the tcl libraries in the tcl install directory next to init.tcl.

I don't use tcl_readline because I run openroad inside emacs that has a real editor. I have never been able to get tcl_readline to compile in my environment.

closed 2021-07-15 by arlpetergadfort