The-OpenROAD-Project/OpenROAD #598: Tests check_buffers, simple_test_clustered, check_wire_rc_cts, check_wire_rc_cts_2 fail with diffs 60.0 um vs. 60 um

yurivict, 2021-03-11:
Differences are like this:
```
*** /usr/ports/cad/openroad/work/OpenROAD-2021_03_09_stable/src/TritonCTS/test/simple_test_clustered.ok Tue Mar  9 10:14:44 2021
--- /usr/ports/cad/openroad/work/OpenROAD-2021_03_09_stable/src/TritonCTS/test/results/simple_test_clustered.log        Thu Mar 11 10:42:42 2021
***************
*** 46,52 ****
   ***********************
  [INFO CTS-0027]  Generating H-Tree topology for net clk
  [INFO CTS-0028]     Tot. number of sinks: 300
! [INFO CTS-0029]     Sinks will be clustered in groups of 10 and a maximum diameter of 60.0 um
  [INFO CTS-0030]     Number of static layers: 1
  [INFO CTS-0020]  Wire segment unit: 14000  dbu (7 um)
  [INFO CTS-0021]  Distance between buffers: 7 units (100 um)
--- 46,52 ----
   ***********************
  [INFO CTS-0027]  Generating H-Tree topology for net clk
  [INFO CTS-0028]     Tot. number of sinks: 300
! [INFO CTS-0029]     Sinks will be clustered in groups of 10 and a maximum diameter of 60 um
  [INFO CTS-0030]     Number of static layers: 1
  [INFO CTS-0020]  Wire segment unit: 14000  dbu (7 um)
  [INFO CTS-0021]  Distance between buffers: 7 units (100 um)
```

mathursanjiv, 2021-03-12:
Where are you seeing this issue? Latest master branch is not giving this issue.
Thanks.

yurivict, 2021-03-12:
I used ```2021_03_09```.

mathursanjiv, 2021-03-12:
Can you try with latest build again? I am not getting any failure. Not sure how this issue might have creeped in.

yurivict, 2021-03-13:
I just tried with the latest master and am still getting this problem.

This might be related to the clang compiler, or to a different c++ library (FreeBSD uses all different libraries compared to Linux).

mathursanjiv, 2021-03-14:
Ok. Is it possible for you to try code change in your environment? If yes, I'll share a change that you can make in code and run.

yurivict, 2021-03-14:
Yes, I can try it.

mathursanjiv, 2021-03-14:
Please try below diff in OpenROAD/src/TritonCTS/src/HTreeBuilder.cpp in function HTreeBuilder::run()

-      _logger->info(CTS, 29, "    Sinks will be clustered in groups of {} and a maximum diameter of {} um",
+      _logger->info(CTS, 29, "    Sinks will be clustered in groups of {} and a maximum diameter of {:.1f} um",

Thanks.

yurivict, 2021-03-14:
This patch fixed the discrepancy.

mathursanjiv, 2021-03-15:
Thanks I'll file PR for this.

closed 2021-03-15 by maliberty