The-OpenROAD-Project/OpenROAD #599: Test gcd fails with a diff in 'Total wirelength'

yurivict, 2021-03-11:
```
*** /usr/ports/cad/openroad/work/OpenROAD-2021_03_09_stable/src/FastRoute/test/gcd.ok   Tue Mar  9 10:14:44 2021
--- /usr/ports/cad/openroad/work/OpenROAD-2021_03_09_stable/src/FastRoute/test/results/gcd.log  Thu Mar 11 10:48:44 2021
***************
*** 34,39 ****
  [INFO GRT-0110] Final usage: 3332
  [INFO GRT-0111] Final number of vias: 1228
  [INFO GRT-0112] Final usage 3D: 7016
! [INFO GRT-0018] Total wirelength: 10266 um
  [INFO GRT-0014] Num routed nets: 563
! No differences found.
--- 34,41 ----
  [INFO GRT-0110] Final usage: 3332
  [INFO GRT-0111] Final number of vias: 1228
  [INFO GRT-0112] Final usage 3D: 7016
! [INFO GRT-0018] Total wirelength: 10264 um
  [INFO GRT-0014] Num routed nets: 563
! Differences found at line 3446.
! 151200 176400 163800 180600 metal1
! 151200 176400 168000 180600 metal1
```

Version: 2021.03.09
OS: FreeBSD 12.2

eder-matheus, 2021-03-11:
@tspyrou @maliberty Please, assign it to me

maliberty, 2021-03-11:
@eder-matheus can't you self-assign?

eder-matheus, 2021-03-11:
@maliberty No, I don't have any permissions in this repo

closed 2021-04-08 by eder-matheus