The-OpenROAD-Project/OpenROAD #604: drt: integer overflow in functions that square lengths

antonblanchard, 2021-03-17:
The llvm undefined behaviour sanitizer is finding integer overflows in tritonroute where it squares lengths. Would it be reasonable to use `frArea`, or do we want a new type, eg:

```
-    frCoord pt2boxDistSquare(const frPoint &pt, const frBox &box);
-    frCoord box2boxDistSquare(const frBox &box1, const frBox &box2, frCoord &dx, frCoord &dy);
-    frCoord box2boxDistSquareNew(const frBox &box1, const frBox &box2, frCoord &dx, frCoord &dy);
+    frArea pt2boxDistSquare(const frPoint &pt, const frBox &box);
+    frArea box2boxDistSquare(const frBox &box1, const frBox &box2, frCoord &dx, frCoord &dy);
+    frArea box2boxDistSquareNew(const frBox &box1, const frBox &box2, frCoord &dx, frCoord &dy);
```

maliberty, 2021-03-17:
It isn't an area so frArea doesn't make sense.  Its more x^2+y^2 than x*y.  

It's hard to come up with a nice name for that though.  Maybe frSquaredDistance ?  I'm open to ideas.

antonblanchard, 2021-03-18:
@maliberty `frSquaredDistance` sounds reasonable to me. I'll work on a patch.

antonblanchard, 2021-03-19:
Fixed in https://github.com/The-OpenROAD-Project/OpenROAD/pull/605

closed 2021-03-19 by antonblanchard