The-OpenROAD-Project/OpenROAD #606: dpl: padding applied to sky130hd endcap, causes placement errors

antonblanchard, 2021-03-19:
I have a sky130hd design that is failing detailed placement. Lots of:
```
 PHY_81117 overlaps PHY_1749
```

There are well taps placed one site before the endcaps:

```
    - PHY_181 sky130_fd_sc_hd__decap_4 + FIXED ( 2503780 255680 ) S ;
    - PHY_10165 sky130_fd_sc_hd__tapvpwrvgnd_1 + FIXED ( 2503320 255680 ) FS ;
```

This seems fine to me. I see `isOverlapPadded` -> `isCrWtBlClass` is meant to handle this, but the endcap cells are just marked as core in the lef file:

```
MACRO sky130_fd_sc_hd__decap_4
  CLASS CORE ;
```

How best to fix this? I thought about marking the decap cells as `ENDCAP`, but if they are just capacitors they aren't limited to being endcaps. Testcase at: [sky130-endcap-padding.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/6173752/sky130-endcap-padding.tar.gz)

maliberty, 2021-04-15:
@msaligane @RTimothyEdwards do you understand what is happening with this cell?  A decap and an endcap are usually different.

RTimothyEdwards, 2021-04-15:
@antonblanchard :  Seems fine to me, too, although I'd like to see specifically the DEF COMPONENTS line for the PHY_81117 and PHY_1749 in the error message.  The ones you cited, PHY_181 and PHY_10165, appear to have the right relative placement, although that's assuming that the DEF units are nanometers (i.e., the tap cell is 0.46um wide and the decap cell is 460 units ahead of the tap, which is fine as long as the DEF file is defining UNITS as 1000 per micron).

I am assuming that this has nothing to do with endcaps, or that you are using "endcap" in the general sense of fill cells placed at the ends of rows to keep all the rows at the same length.  LEF macros have a very specific sense of an ENDCAP being a cell that connects the end of a row to power rails, or to otherwise close off a row for some DRC related purpose like ensuring sufficient deep nwell surrounding the digital core, or something like that.  There are no such endcap requirements in the SkyWater standard cell library so it would not be appropriate to designate any core cell as ENDCAP.

RTimothyEdwards, 2021-04-15:
@antonblanchard :  For what it's worth, I pulled the gzipped file and read in the layout, and there is nothing wrong with it.  It appears that the tools are confused about what constitutes an overlap. . .

maliberty, 2021-04-15:
@RTimothyEdwards they don't overlap in a drc sense but they do overlap when you consider padding.  Normally we exclude padding on endcaps by looking at their type.  On this decap cell we just see it as a core cell so it expects to be padded.  I don't see any way from the lef that we could determine this or even distinguish it from a decap not used as an endcap.  The best stupid idea I have it consider any cell with only power/gnd pins as zero padding.

RTimothyEdwards, 2021-04-15:
@maliberty :  What is meant by "padding" in this context?

maliberty, 2021-04-15:
@RTimothyEdwards sites left free on the left and/or right to reduce placement density / routing congestion.  In openlane you can see it in or_opendp.tcl:

set_placement_padding -global -right $::env(CELL_PAD)

set_placement_padding -masters $::env(CELL_PAD_EXCLUDE) -right 0 -left 0

It looks like you are excluding decaps:
set ::env(CELL_PAD_EXCLUDE) "sky130_fd_sc_hd__tap* sky130_fd_sc_hd__decap* sky130_fd_sc_hd__fill*"

we have no similar exclude in ORFS.

maliberty, 2021-04-15:
@antonblanchard do you want to try tackling this?

@tspyrou an example of flow divergence from OL that hurts us.

RTimothyEdwards, 2021-04-15:
@maliberty :  I have no idea of either why openlane is set up to exclude any cell from being considered for padding.  I can't think of any reason to make such a restriction.  However, I am also confused by why a column of decap has been pre-placed up each side in the tarball example DEF file.

maliberty, 2021-04-15:
@RTimothyEdwards because the decap is specified as an endcap.  We could add the ability to have no endcap cell and only put well taps if that is appropriate.

RTimothyEdwards, 2021-04-15:
@maliberty :  My understanding from what you just said is that "padding" is defined as a certain number of sites on the right and possibly the left that are left empty in the initial placement to allow the place & route tool some margin to spread things out to avoid congestion.  And if you have to have endcaps on every row, then they would have to be preplaced and the padding sites would have to clear those.  So my understanding is that the OpenROAD place & route does not clear the endcaps and so that's missing functionality that you need to add?  But the rest of it I don't get.  There is nowhere in the test case tarball that the decap cell is defined to be an ENDCAP.  Maybe that was done when the DEF was created but didn't get into the macro definitions that were included with the test case tarball?  Otherwise how do the decap cells end up in the preplacement DEF?

maliberty, 2021-04-20:
@RTimothyEdwards the decaps are set as endcaps [here](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/platforms/sky130hd/tapcell.tcl)

tapcell \
  -endcap_cpp "2" \
  -distance 14 \
  -tapcell_master "sky130_fd_sc_hd__tapvpwrvgnd_1" \
  -endcap_master "sky130_fd_sc_hd__decap_4"

I've discussed with @eder-matheus making endcap_master optional if the platform doesn't require endcaps.

maliberty, 2021-04-20:
There is no need for padding of decaps as they have no non-power terminals and mostly act like a filler anyways.

maliberty, 2021-05-05:
I believe this is resolved; @antonblanchard would you confirm and close.

closed 2021-07-06 by jjcherry56