The-OpenROAD-Project/OpenROAD #609: SEGV in Resizer::repairSetup(), passing negative value to expanded.path()

antonblanchard, 2021-03-24:
The attached test case gets a SEGV when run with the latest OpenROAD. We are passing a negative value into `expanded.path()`:

```
--- a/src/rsz/src/Resizer.cc
+++ b/src/rsz/src/Resizer.cc
@@ -2052,6 +2052,10 @@ Resizer::repairSetup(PathRef &path,
       LibertyPort *in_port = network_->libertyPort(in_pin);
 
       int prev_drvr_index = drvr_index - 2;
+      if (prev_drvr_index < 0) {
+             fprintf(stderr, "prev_drvr_index bad\n");
+             exit(1);
+      }
       PathRef *prev_drvr_path = expanded.path(prev_drvr_index);
```

Test case: [repairSetup-segv.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/6193589/repairSetup-segv.tar.gz)

jjcherry56, 2021-03-24:
fixed in ab5e4af3 rsz repair setup paths thru latches (issue #609)

closed 2021-03-24 by jjcherry56

antonblanchard, 2021-03-25:
Thanks @jjcherry56