The-OpenROAD-Project/OpenROAD #611: CTS post processing makes clock skew significantly worse

antonblanchard, 2021-03-25:
I'm noticing high clock skew on some `sky130hd` designs. While trying to narrow this down, I found that adding `-post_cts_disable` helps a lot:

```
baseline (cts.tcl):
Clock ext_clk
Latency      CRPR       Skew
_108993_/CLK ^
   6.34
_108913_/CLK ^
   5.39      0.00       0.95
```

```
-post_cts_disable (cts-disable.tcl):
Clock ext_clk
Latency      CRPR       Skew
_108334_/CLK ^
   5.84
_108543_/CLK ^
   5.39      0.00       0.45
```

It seems like CTS post processing is adding quite a few buffers to a few clock nets and making things worse overall. Note that the design in question has a very low utilisation because it is missing some macros that need to be placed (not sure it matters, but I thought I should mention it).

Test case: [cts-issue.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/6204234/cts-issue.tar.gz)

antonblanchard, 2021-04-02:
It looks like this optimisation just adds buffers if the wire length is significantly larger (5x) than the average. That seems overly simplistic. Can we call into STA to get a better idea of the delays on every clock net, and drive the skew down?

It could also detect if we have skew or capacitance violations on the clock nets, and fix them. I see that quite a lot, and have been struggling to fix it in https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/10 and https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/9 

Note: I'm going on the presumption that we aren't inserting beneficial slack in the clock tree to help meet timing. Is that correct?

mathursanjiv, 2021-04-02:
> to fix it in The-OpenROAD-Project/OpenROAD-flow-scripts#10
> and The-OpenROAD-Project/OpenROAD-flow-scripts#9

Yes, the plan is to do STA based buffering and get rid of characterization
based buffering post CTS. This also required bringing in parasitic
estimator and running STA as part of CTS. There is no useful skew
implementation yet.

vijayank88, 2023-06-14:
with latest commit test case `cts.tcl` generated skew:
```
post cts report_clock_skew
--------------------------------------------------------------------------
Clock ext_clk
Latency      CRPR       Skew
_103560_/CLK ^
   3.29
_103751_/CLK ^
   2.80      0.00       0.48
```
With `cts-disable.tcl` also produces same skew.
```
post cts report_clock_skew
--------------------------------------------------------------------------
Clock ext_clk
Latency      CRPR       Skew
_103560_/CLK ^
   3.29
_103751_/CLK ^
   2.80      0.00       0.48
```
@antonblanchard 
Is required any action on this still?

maliberty, 2025-08-30:
There have been substantial changes since this was filed.  Please reopen if still relevant