The-OpenROAD-Project/OpenROAD #614: Unable to install OpenRoad using Docker

vivekkrajan, 2021-03-31:
Hi - 

I am trying to install OpenRoad using Docker Image and getting errors. 

https://theopenroadproject.org/2019/12/11/getting-started-with-openroad-app-part-2/

docker pull openroad/openroad:base
git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD.git
cd OpenRoad
docker run -it -B $(pwd):/OpenROAD docker://docker.io/openroad/openroad:base bash

cd /OpenROAD
mkdir build
cd build
cmake ..

Getting the error - see image. 

[image]

Please help. Please release an image that addresses this issue. 

Thanks,
Vivek Rajan

vvbandeira, 2021-04-01:
That post you reference from 2019 is out-of-date.

If you want to use a docker image to build I suggest one of the images below depending on your OS preference.

https://hub.docker.com/r/openroad/centos7-dev
https://hub.docker.com/r/openroad/ubuntu20-dev

If you want, there's the docker helper script `./etc/DockerHelper.sh` with it you can build, test, or run the app with simpler commands. See `./etc/DockerHelper.sh -help` for more information on the usage and features available.

mithro, 2021-04-01:
There is also work by the @umarcor at https://github.com/hdl/containers to provide optimized OpenROAD + OpenLane docker containers.

vvbandeira, 2021-04-05:
@vivekkrajan were you able to use either the images or the helper script to compile the app?

vivekkrajan, 2021-04-05:
Thanks for follow-up.
I have not been able to install OpenRoad yet. Looking forward for help and resolving the issues.

vvbandeira, 2021-04-05:
> That post you reference from 2019 is out-of-date.
>
> If you want to use a docker image to build I suggest one of the images below depending on your OS preference. [...]

Did you try to use any of the images I mentioned? What was the outcome?

Did you try to use the scripts I mentioned? What was the outcome?

If you have the output with error, paste the complete output on https://pastebin.pl/ or similar site so we can look.

peijunh, 2021-06-15:
It seems that using the latest version will give this error?

CMake Error in /OpenROAD/build/CMakeFiles/CMakeTmp/CMakeLists.txt:
  Target "cmTC_2ef56" requires the language dialect "CXX17" , but CMake does
  not know the compile flags to use to enable it.

CMake Error at /usr/local/share/cmake-3.14/Modules/CheckIncludeFileCXX.cmake:92 (try_compile):
  Failed to generate test project build system.
Call Stack (most recent call first):
  /usr/local/share/cmake-3.14/Modules/FindThreads.cmake:110 (CHECK_INCLUDE_FILE_CXX)
  /usr/local/lib64/cmake/spdlog/spdlogConfig.cmake:4 (find_package)
  src/CMakeLists.txt:107 (find_package)

My comamand is here:

docker pull openroad/centos7-dev:latest
git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD.git
cd OpenROAD
docker run -it -v $(pwd):/OpenROAD openroad/centos7-dev:latest bash
[root@cdcfacbddb47 /]# cd OpenROAD/
[root@cdcfacbddb47 OpenROAD]# mkdir build
[root@cdcfacbddb47 OpenROAD]# cd build
[root@cdcfacbddb47 build]# cmake ..

maliberty, 2021-06-15:
You need:  source /opt/rh/devtoolset-8/enable

peijunh, 2021-06-15:
> You need: source /opt/rh/devtoolset-8/enable
> 
> (See etc/Build.sh)

Do I need to do it before docker run or after?

maliberty, 2021-06-15:
after (inside the container)

peijunh, 2021-06-15:
> after (inside the container)

Thanks for your reply, but it seems that there is still error.

CMake Error at src/CMakeLists.txt:118 (add_subdirectory):
  The source directory

    /OpenROAD/src/sta

  does not contain a CMakeLists.txt file.

-- The C compiler identification is GNU 8.3.1
-- Check for working C compiler: /opt/rh/devtoolset-8/root/usr/bin/cc
-- Check for working C compiler: /opt/rh/devtoolset-8/root/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- GUI is enabled
-- Boost 1.68.0 found.
-- Found OpenMP_CXX: -fopenmp (found version "4.5")
-- Found OpenMP: TRUE (found version "4.5")
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS)
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring incomplete, errors occurred!

I run source /opt/rh/devtoolset-8/enable before I run cmake .. and then get this. All the docker and OpenROAD are newly pulled.

vvbandeira, 2021-06-15:
@peijunh did you remove the old build directory?

Also, there's a build script to minimize config/setup errors. Can you try to use it instead of manually calling `cmake` and `make`?
```
./etc/Build.sh -clean
```

peijunh, 2021-06-15:
@maliberty @vvbandeira Thank you for the help, the docker sucessfully build the openroad. And the command openroad --version gives feedback. But when I quit the openroad docker container, the openroad command becomes unknown. So should I always use docker image to run openroad? or the openroad is actually installed in my system but I just don't know?

vvbandeira, 2021-06-15:
@peijunh the `docker run` command creates a container (similar to a VM), everything you run/use/install after `docker run` is isolated from your local system (unless you mount a volume), thus openroad not being available outside docker is the expected behavior.

vvbandeira, 2021-07-24:
@vivekkrajan any updates? Were you able to use docker?

vivekkrajan, 2021-07-24:
Thanks, Yes, able to make progress. This issue is closed.

closed 2021-07-24 by vvbandeira