The-OpenROAD-Project/OpenROAD #615: Docker install fails on OSX due to make options

KatieTheMathematician, 2021-04-02:
OSX does not support build -j or the nproc command. sysctl -n hw.logicalcpu works in place of nproc as does deleting the -j options from make/build commands. A (potentially suboptimal) combination of edits that resulted in a successful docker install while otherwise following the tutorial on Big Sur 11.2.3 is below. 

Note: git diff run from OpenROAD-flow-scripts/
```
diff --git a/build_openroad.sh b/build_openroad.sh
--- a/build_openroad.sh
+++ b/build_openroad.sh
@@ -20,7 +20,7 @@ function usage() {
 
 # defaults
 NICE=""
-PROC=`nproc`
+PROC=`sysctl -n hw.logicalcpu`
 
 # Parse arguments
 while (( "$#" )); do
@@ -119,10 +119,10 @@ if [ "$build_method" == "DOCKER" ]; then
 # Local build
 elif [ "$build_method" == "LOCAL" ]; then
   mkdir -p tools/build/yosys
-  (cd tools/yosys && $NICE make install -j$PROC PREFIX=../build/yosys CONFIG=gcc)
+  (cd tools/yosys && make install PREFIX=../build/yosys CONFIG=gcc)
 
   mkdir -p tools/build/OpenROAD
-  (cd tools/build/OpenROAD && cmake ../../OpenROAD && $NICE make -j$PROC)
+  (cd tools/build/OpenROAD && cmake ../../OpenROAD && make)
 else
   echo "ERROR: No valid build method found"
   exit 1
```
Note: git diff run from OpenROAD-flow-scripts/
```
diff --git a/tools/yosys_util/Dockerfile b/tools/yosys_util/Dockerfile
--- a/tools/yosys_util/Dockerfile
+++ b/tools/yosys_util/Dockerfile
@@ -20,9 +20,9 @@ RUN yum install -y flex tcl tcl-devel libffi-devel git graphviz readline-devel g
     tar -xvzf bison-3.0.1.tar.gz && \
     cd bison-3.0.1 && \
     ./configure && \
-    make -j$(nproc) && \
+    make && \
     make install
 
 COPY . /yosys
 WORKDIR /yosys
-RUN make PREFIX=/build CONFIG=gcc install -j$(nproc)
+RUN make PREFIX=/build CONFIG=gcc install
```
Note: git diff run from OpenROAD-flow-scripts/tools/OpenROAD/
```
diff --git a/jenkins/docker/Dockerfile.centos7.gcc8 b/jenkins/docker/Dockerfile.centos7.gcc8
--- a/jenkins/docker/Dockerfile.centos7.gcc8
+++ b/jenkins/docker/Dockerfile.centos7.gcc8
@@ -96,4 +96,4 @@ COPY . /OpenROAD
 WORKDIR /OpenROAD
 
 # Build
-RUN cmake -B build . && cmake --build build -j 8
+RUN cmake -B build . && cmake --build build

```

maliberty, 2021-04-02:
Openlane does:

ifeq (, $(strip $(NPROC)))
  # Linux (utility program)
  NPROC := $(shell nproc 2>/dev/null)

  ifeq (, $(strip $(NPROC)))
    # Linux (generic)
    NPROC := $(shell grep -c ^processor /proc/cpuinfo 2>/dev/null)
  endif
  ifeq (, $(strip $(NPROC)))
    # BSD (at least FreeBSD and Mac OSX)
    NPROC := $(shell sysctl -n hw.ncpu 2>/dev/null)
  endif
  ifeq (, $(strip $(NPROC)))
    # Fallback
    NPROC := 1
  endif

endif

vvbandeira, 2021-04-02:
Although we do not officially support MacOS, it is know to work on it (with the correct packages installed).
Have you installed `coreutils`? `brew install coreutils`
This will install `nproc`, just make sure your path variable is updated.

The flag `-j ` from make should not be an issue if you are building with Docker, since these commands would only be called from inside a docker container.

KatieTheMathematician, 2021-04-02:
Big Sur seems to work quite well with the Docker build despite the lack of official support. I also have a stand alone Yosys install that was not difficult to get running. I don't have much experience with Docker so I'm not sure why -j is making a difference. I've included the error messages if that helps.

On re-pulling the repo and rebuilding after installing coreutils the nproc issue was fixed in the main build script but the other two still occurred.

Yosys produced the below error:

```
#11 100.4 gcc: internal compiler error: Killed (program cc1plus)
#11 100.4 Please submit a full bug report,
#11 100.4 with preprocessed source if appropriate.
#11 100.4 See <http://bugzilla.redhat.com/bugzilla> for instructions.
#11 100.4 make: *** [Makefile:688: passes/opt/opt_expr.o] Error 4
#11 100.4 make: *** Waiting for unfinished jobs....
#11 104.2 passes/techmap/abc.cc: In member function 'virtual void {anonymous}::AbcPass::execute(std::vector<std::basic_string<char> >, Yosys::RTLIL::Design*)':
#11 104.2 passes/techmap/abc.cc:1815:15: warning: 'g_argidx' may be used uninitialized in this function [-Wmaybe-uninitialized]
#11 104.2       cmd_error(args, g_argidx, stringf("Unsupported gate type: %s", g.c_str()));
#11 104.2       ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
------
executor failed running [/bin/sh -c make PREFIX=/build CONFIG=gcc install -j$(nproc)]: exit code: 2
```
It ran successfully after simply deleting the call to nproc from my 2nd git diff.
```
-RUN make PREFIX=/build CONFIG=gcc install -j$(nproc)
+RUN make PREFIX=/build CONFIG=gcc install
```
I also still needed to remove the -j option in the 3rd git diff. The error message is similar.

```
#18 287.7 [ 44%] Built target OpenSTA
#18 287.7 gmake[1]: Leaving directory '/OpenROAD/build'
#18 287.7 gmake: *** [Makefile:133: all] Error 2
------
executor failed running [/bin/sh -c cmake -B build . && cmake --build build -j 8]: exit code: 2
```

vvbandeira, 2021-04-02:
I don't know what is causing this error and I can't reproduce.

Docker version 20.10.5, build 55c4c88
macOS BigSur 11.2.3 (20D91)

```
git clone https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts flow
cd flow
./build_openroad.sh
```

Works without issue.

vvbandeira, 2021-04-05:
@KatieTheMathematician were you able to compile? If not could you run the following commands and upload the `output.txt` file generated into https://pastebin.pl/ or a similar site?

```
git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts issue-615
cd issue-615
./build_openroad.sh &> output.txt
```

KatieTheMathematician, 2021-04-07:
I was perhaps a bit hasty in concluding what was wrong with the -j option in the Yosys Dockerfile. You were correct in your initial assessment that it was unrelated to running Docker from an OSX host. If I hardcode it to use one core it builds. From the 2nd git diff change RUN make PREFIX=/build CONFIG=gcc install -j$(nproc) to RUN make PREFIX=/build CONFIG=gcc install -j1 

I haven't tried the same thing for the OpenSTA container due to the build time but I'm relatively certain the outcome will be the same. 

It may be one of those hard to nail down things that indicates the build order is a bit too permissive. (See https://stackoverflow.com/questions/1564195/gnu-makes-j-option .) I'm attaching the output from the failed Yosys/OpenSTA builds to hopefully help you figure out what is happening out of order assuming this is the problem.  

[OpenSTA_build_fail.txt](https://github.com/The-OpenROAD-Project/OpenROAD/files/6268820/OpenSTA_build_fail.txt)

[yosys_build_fail.txt](https://github.com/The-OpenROAD-Project/OpenROAD/files/6268810/yosys_build_fail.txt)

vvbandeira, 2021-04-07:
I don't think this is related to make or a build issue since it does not occur on our CI and I cannot reproduce locally even after many tries on my macOS machine.

Also, the log files you provided are not complete.

KatieTheMathematician, 2021-04-07:
I have a solution! This was a weird one and I'm not sure if there's anything the OpenROAD team would do to fix it but hopefully this will help anyone who stumbles on the issue in the future:
My docker container was running out of memory which was crashing GCC. The relevant error message was the one from Yosys. "gcc: internal compiler error: Killed (program cc1plus)" This indicates GCC crashed. This is frequently, but not necessarily a resource issue as GCC is quite stable. I had installed Docker without changing any of the defaults. Increasing its memory to 8GB, CPU cores to 8, and swap to 2GB allowed me to build without issue.  This also explains why reducing the number of threads for build allowed me to run but you weren't seeing any issues with your CI pipeline. Thank you very much for assisting me with walking through the issue.

vvbandeira, 2021-04-07:
No problem, happy to help and thanks for posting the root cause and solution you found.

Please mark the issue as closed when you get a chance.

Regards,
--
Vitor Bandeira

maliberty, 2021-04-07:
@vvbandeira perhaps we can add a note about minimum docker requirements in the readme.

closed 2021-04-07 by KatieTheMathematician