The-OpenROAD-Project/OpenROAD #616: regression tests in /test.

ripudamank2, 2021-04-02:
Hi, I was running regression test and the below errors occurred.
What should i do now to fix this problem .

xxx@xxx:~/Documents/OpenROAD/test$ ./regression flow
------------------------------------------------------
Flow
gcd_nangate45 *ERROR* segmentation violation
aes_nangate45 *ERROR* segmentation violation
tinyRocket_nangate45 *ERROR* segmentation violation
gcd_sky130hs *ERROR* segmentation violation
aes_sky130hs *ERROR* segmentation violation
ibex_sky130hs *ERROR* segmentation violation
jpeg_sky130hs *ERROR* segmentation violation
gcd_sky130hd *ERROR* segmentation violation
aes_sky130hd *ERROR* segmentation violation
ibex_sky130hd *ERROR* segmentation violation
jpeg_sky130hd *ERROR* segmentation violation
------------------------------------------------------
Errored 11/11
See /home/xxx/Documents/OpenROAD/test/results for log files
------------------------------------------------------
Found 1 tool regression failures.

maliberty, 2021-04-03:
Start by running one design (eg gcd) and see where it is crashing.  We don't see this in our nightly regression so I'm not sure what the issue is.

ripudamank2, 2021-04-04:
i am new with OpenROAD , can you please tells some steps how to run one design .

ripudamank2, 2021-04-04:
[gcd_nand45_error.txt](https://github.com/The-OpenROAD-Project/OpenROAD/files/6254810/gcd_nand45_error.txt)

I tried to run like this **openroad gcd_nangate45.tcl**  but i got the error.
Can you please look into that.

maliberty, 2021-04-05:
How did you compile openroad?  On what platform?  With what compiler?  Are you using Docker?  This isn't enough to be actionable.

ripudamank2, 2021-04-05:
​$ mkdir build
$ cd build
$ cmake ..
$ make

I use the above steps which you have mentioned in github link .

I am using latest stable ubuntu 20.04 with gcc 9

vijayank88, 2021-11-30:
@ripudamank2  are you still facing same issue?

vvbandeira, 2022-11-28:
Closing due to lack of activity. Please open a new issue in case the problem is still relevant and occurring with the latest commit.

closed 2022-11-28 by vvbandeira