The-OpenROAD-Project/OpenROAD #617: Install stuck at 73%

vivekkrajan, 2021-04-02:
Hi - 

I am trying to install OpenRoad and not able to make much progress, it is stuck at 73%, see picture below. 

This is happening after:
  cd /OpenROAD
  mkdir build
  cd build
  cmake ..
  make -j 4 ---> Picture below is from running this command. I am running on a large machine 128G with 4 Cores 

[image]

Appreciate any BKMs and suggestions to resolve this issue. 
What is happening at 73% of compile? 

Thank you,
Vivek Rajan

maliberty, 2021-04-02:
Can you run -j1 to see exactly which file it is getting stuck on.  I never see this problem so it seems like something local.

vivekkrajan, 2021-04-02:
Thanks for your help. I re-ran with "-j 1" It is now stuck in 52%. 

Any idea what I am missing? 

[image]

Is there step-by-step instruction on how to install OpenRoad ..? Please share latest pointers. 

Thanks,
Vivek Rajan

maliberty, 2021-04-03:
@vvbandeira Are there update to date instructions?  I think these might be a bit old: https://openroad.readthedocs.io/en/latest/user/GettingStarted.html

ripudamank2, 2021-04-04:
Had you tried this step :
uncomment the export_opendb_power_pins in proc opendb_update_grid {} in OpenROAD/src/pdngen/src/PdnGen.tcl and then rebuild the OpenROAD tool.

vvbandeira, 2021-04-05:
Could you run the following commands and upload the `output.txt` file generated into https://pastebin.pl/ or a similar site?

```
git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD issue-617
cd issue-617
./etc/Build.sh &> output.txt
```

or if you have special compiler paths / environment

```
git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD issue-617
cd issue-617
mkdir build
cd build
cmake .. &> output.txt
make -j 1  &>> output.txt
```

maliberty, 2021-04-20:
@ripudamank2 is this still an issue or can we close this?

vivekkrajan, 2021-04-21:
Please close this issue. A colleague of mine was able to install OpenRoad. Thanks.

closed 2021-04-21 by maliberty