The-OpenROAD-Project/OpenROAD #619: Report congestion map in global route

vz-rocks, 2021-04-05:
There was this argument of "report_congestion" in global_route function but it is gone now. What is the motivation to remove that feature and is there any way to output the congestion map now? Thanks.

eder-matheus, 2021-04-05:
@vz-rocks The "report_congestion" argument generates a large text file with the information of all GCells. It was really hard to understand. Now we have the congestion map in the GUI, which is way better to understand where the congestion is happening.

eder-matheus, 2021-04-14:
@vz-rocks Is there anything that you would like to be updated in the global_route regarding the congestion report? If not, please close the issue :)

maliberty, 2021-04-20:
This has gone silent so closing.

closed 2021-04-20 by maliberty