The-OpenROAD-Project/OpenROAD #621: SEGV in grt::orderx()

antonblanchard, 2021-04-09:
The following test case hits a SEGV in grt::orderx():

```
#0  0x0000000000e44920 in grt::orderx(grt::pnt const*, grt::pnt const*) ()
#1  0x0000000000e478c8 in void std::__introsort_loop<__gnu_cxx::__normal_iterator<grt::pnt**, std::vector<grt::pnt*, std::allocator<grt::pnt*> > >, long, __gnu_cxx::__ops::_Iter_comp_iter<int (*)(grt::pnt const*, grt::pnt const*)> >(__gnu_cxx::__normal_iterator<grt::pnt**, std::vector<grt::pnt*, std::allocator<grt::pnt*> > >, __gnu_cxx::__normal_iterator<grt::pnt**, std::vector<grt::pnt*, std::allocator<grt::pnt*> > >, long, __gnu_cxx::__ops::_Iter_comp_iter<int (*)(grt::pnt const*, grt::pnt const*)>) ()
#2  0x0000000000e462e8 in grt::fluteNormal(int, int, int*, int*, int, float, grt::Tree*) ()
#3  0x0000000000e46a38 in grt::gen_brk_RSMT(char, char, char, char, char, utl::Logger*) ()
#4  0x0000000000e3d22e in grt::FastRouteCore::run() ()
#5  0x0000000000af2044 in grt::GlobalRouter::findRouting(std::vector<grt::Net*, std::allocator<grt::Net*> >&, int, int) ()
#6  0x0000000000af75ce in grt::GlobalRouter::globalRoute() ()
#7  0x00000000004c2452 in gpl::RouteBase::getGlobalRouterResult() ()
#8  0x00000000004c6069 in gpl::RouteBase::routability() ()
#9  0x00000000004a36fc in gpl::NesterovPlace::doNesterovPlace() ()
#10 0x00000000004ce450 in _wrap_replace_nesterov_place_cmd ()
```

It bisects to:

```
commit 1557cb0f3526fce41628b6cd9dd8f6f1d6419b27 (HEAD)
Author: eder-matheus <emrmonteiro@inf.ufrgs.br>
Date:   Wed Apr 7 20:43:34 2021 -0300

    grt: use vector and std::sort for ptp in RSMT.cpp
```

Test case: 
[gpl_segv.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/6282383/gpl_segv.tar.gz)

maliberty, 2021-04-09:
There was a another recent change here that might fix this.

eder-matheus, 2021-04-09:
> There was a another recent change here that might fix this.

Yes, my changes right after the last commit to the public repo fix this issue. I just tested the .tar in my machine with the latest version of the private repo and it don't crash anymore.

eder-matheus, 2021-04-14:
@antonblanchard The fix for this issue is already available in the repository. Can you check it?

maliberty, 2021-04-20:
@antonblanchard is this resolved for you?

eder-matheus, 2021-04-27:
@antonblanchard @maliberty I will close the issue since I was able to run the testcase with no errors using the latest commit of the public repo.

closed 2021-04-27 by eder-matheus