The-OpenROAD-Project/OpenROAD #623: Support Verilog AMS

handicraftsman, 2021-04-11:
This is probably an epic, but still.

I guess that two primary uses for OpenROAD are education (possibly self-education like in my case) and open source hardware.

If it is so, this flow misses support for analog-mixed-signal cores which are critical for powering actual devices and interacting with many peripherals.

Asking this here because currently there are *no* open source Verilof/VHDL AMS supporting flows.

maliberty, 2021-04-11:
I agree with the sentiment but this is too far outside the bounds of anything we are planning to work.  The closest project I know of is https://fasoc.engin.umich.edu/

closed 2021-04-11 by maliberty

handicraftsman, 2021-04-12:
Thanks, this seems nice.