The-OpenROAD-Project/OpenROAD #624: script for OpenRCX reference extractor

Michael-JY-He, 2021-04-18:
Could you please provide an example script for the OpenRCX reference extractor? 
To be more specific, I'm trying to use Cadence Innovus to extract parasitics. I'm having problem in loading .def or .v into Innovus. 

The Innovus user guide says:
You cannot use the DEF netlistflow for parasitic extraction, delay calculation, and timing-driven placement androuting effectively because the DEF names do not properly match the Verilog namesused in timing constraints and timing analysis.

maliberty, 2021-04-20:
You can see the flow script usage [here](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/efbbca443cf07a3f5f2d06a6dfcd8751a29a6ce5/flow/scripts/final_report.tcl#L41)

Michael-JY-He, 2021-04-20:
I guess what I'm try to ask is, how to run the [step B](https://github.com/The-OpenROAD-Project/OpenRCX/blob/a0c988ddc591b0b6e18c073d72a112aeb81a4bc3/doc/calibration.txt#L15) because the script is not provided.

maliberty, 2021-04-20:
We cannot provide scripts for commercial tools as they are covered by NDA.  @GeraldoPradipta have you seen such issues in your testing?

GeraldoPradipta, 2021-04-20:
No, I haven't had any problem with loading the DEF and the verilog files to Innvous.

@Michael-JY-He, what issues you have exactly? Is it feeding the DEF and Verilog to Innovus? Or the DEF and the verilog do not match?

vijayank88, 2023-06-14:
@Michael-JY-He 
Is this resolved?

rovinski, 2023-06-14:
@maliberty interesting question - could we possibly include commercial scripts now given that Cadence allows sharing of scripts with a disclaimer? Maybe if not in this repo then another one.

maliberty, 2023-06-14:
@tspyrou any thoughts on releasing such scripts?

Michael-JY-He, 2023-06-19:
I'm not in the project now. Let me close the issue.

closed 2023-06-19 by Michael-JY-He