The-OpenROAD-Project/OpenROAD #628: 'Not enough memory' error at detailed routing step

stevvik, 2021-05-04:
I'm trying to run test design through OpenROAD flow. It is a simple combinational scheme that contains ~4K gates. For this number of gates I have to increase die area up to 1000x1000 um. With this area design passes successfully. Buf if I try to increase die area up to ~2000x2000um, the openroad flow fails at the begining of detailed routing step with the following error:

set routed_def [make_resultfile ${design}${platform}_route.def]
set tr_lef [make_tr_lef]
Error: flow.tcl, 149 couldn't fork child process: not enough memory

I run openroad in container at host with 8G memory.

According to Matt Liberty recomendatin in gitter community I opened this issue and created a link to design.
Link below contains tuned dir structure with failed design, runfiles and logs. RUN.log contains log of failed run.
Design is created from gcd test example from openroad flow. I had to comment CTS step in flow.tcl to avoid errors that clock is absent in design.

[https://drive.google.com/file/d/1_Yz6AcxzRaMImOt5hrs5_20ks-IXL2uX/view?usp=sharing](url)

maliberty, 2021-05-04:
I tweaked your test case to run with the latest openroad and I can see the virtual size of openroad exceed 8Gb.  However I would expect that would just lead to swapping no an out of memory error.  Do you have swapping enable / have sufficient swap inside your container?

maliberty, 2021-05-04:
You do not have 4k instances, you have 116,189 instances.  There are 127 instances of crc32_table * 883 std cell per instace = 112,141 instances plus some 4k top level std cells.

Resizer will add some buffers too.  Increasing the die area increases the fill cell count so you get to 809,675 instances by the time detailed routing starts with 126k nets to route.

For this scenario the memory usage seems reasonable.

closed 2021-05-04 by maliberty

stevvik, 2021-05-04:
Matt, thank you for answer! Where did you take this statistic for instances? I got 4K by summing all gates after yosys (near the end of its log). I sow a big number of instances in openroad log, but decided that it is not a number of gates, but a total number of placed instances, for example, including vias. 

— Евгений Столбиков

maliberty, 2021-05-04:
We print the number at various points in the flow but you can always do:

set blk [ord::get_db_block]
puts "Num insts = [llen [$blk getInsts]]"

stevvik, 2021-05-04:
Thank you. And what are the ‘instances’ in your case? Are they a standard cells only or another topology figures too?

— Евгений Столбиков

maliberty, 2021-05-04:
placeable instances - standard cells or macros or fill cells.  Basically anything that would appears in the COMPONENTS section in DEF.