The-OpenROAD-Project/OpenROAD #629: dl.bintray.com 403 forbidden

Jenny-Tan, 2021-05-06:
Operating System: Ubuntu 20.04

When I run
```
./etc/DependencyInstaller.sh -dev
```
I got the following message:
```
--2021-05-06 10:21:44--  https://dl.bintray.com/boostorg/release/1.72.0/source/boost_1_72_0.tar.gz
Resolving dl.bintray.com (dl.bintray.com)... 34.213.218.48, 52.12.180.214, 52.40.49.72, ...
Connecting to dl.bintray.com (dl.bintray.com)|34.213.218.48|:443... connected.
HTTP request sent, awaiting response... 403 Forbidden
2021-05-06 10:21:45 ERROR 403: Forbidden.
```
It seems that this website is down.

After I run 
```
cmake ..
```

I got the following message
```
-- OpenROAD version: 1
-- OpenROAD git sha: fca18d9c7f54f67a79d05c7958def4068b0255db
-- System name: Linux
-- Compiler: GNU 9.3.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- TCL readline library: /usr/lib/x86_64-linux-gnu/libtclreadline.so
-- TCL readline header: /usr/include/x86_64-linux-gnu
CMake Error at /usr/local/share/cmake-3.14/Modules/FindBoost.cmake:2132 (message):
  Unable to find the requested Boost libraries.

  Unable to find the Boost header files.  Please set BOOST_ROOT to the root
  directory containing Boost or BOOST_INCLUDEDIR to the directory containing
  Boost's headers.
Call Stack (most recent call first):
  src/CMakeLists.txt:71 (find_package)

CMake Error at src/CMakeLists.txt:109 (find_package):
  By not providing "Findspdlog.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "spdlog", but
  CMake did not find one.

  Could not find a package configuration file provided by "spdlog" with any
  of the following names:

    spdlogConfig.cmake
    spdlog-config.cmake

  Add the installation prefix of "spdlog" to CMAKE_PREFIX_PATH or set
  "spdlog_DIR" to a directory containing one of the above files.  If "spdlog"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/home/tyww/Desktop/OpenROAD/build/CMakeFiles/CMakeOutput.log".
```
Is there a way to properly install all the dependencies? Thank you.

maliberty, 2021-05-06:
You have an old version, this is already fixed, see

https://github.com/The-OpenROAD-Project/OpenROAD/blob/2d4e8c5d8d3422dfb8888013e7a088b2a445df03/etc/DependencyInstaller.sh#L39

Jenny-Tan, 2021-05-06:
Thank you for your response!

After I tried the new version, I still cannot install the boost library.

```
./bootstrap.sh: 1: ./tools/build/src/engine/build.sh: not found
Building Boost.Build engine with toolset ... 
Failed to build Boost.Build build engine
Consult 'bootstrap.log' for more details
```

It turns out that the `./tool/build` is an empty directory. I tried to fix this problem by downloading the `build` directory and rebuild the boost. Then I found that many of the other directories for building the boost is empty.

I wonder what should I do with this problem? Thank you!

maliberty, 2021-05-06:
We build our docker containers this way so I am not sure what is different in your setup.  @vvbandeira any thoughts on this?

Jenny-Tan, 2021-05-25:
I got the following error when running ``sudo ./etc/DependencyInstaller.sh -dev``
```Installation complete
./etc/DependencyInstaller.sh: line 41: cd: boost-1.72.0: Not a directory
```
Then when I run ``cmake ..`` I got
```
-- The CXX compiler identification is GNU 9.3.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: 1
-- OpenROAD git sha: 130274741ed9d0731cb08828665e8c722a6af630
-- System name: Linux
-- Compiler: GNU 9.3.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- TCL readline library: /usr/lib/x86_64-linux-gnu/libtclreadline.so
-- TCL readline header: /usr/include/x86_64-linux-gnu
-- Found SWIG: /usr/bin/swig (found suitable version "4.0.1", minimum required is "3.0") 
CMake Error at /usr/local/share/cmake-3.14/Modules/FindBoost.cmake:2132 (message):
  Unable to find the requested Boost libraries.

  Unable to find the Boost header files.  Please set BOOST_ROOT to the root
  directory containing Boost or BOOST_INCLUDEDIR to the directory containing
  Boost's headers.
Call Stack (most recent call first):
  src/CMakeLists.txt:70 (find_package)

-- Found Python3: /usr/lib/x86_64-linux-gnu/libpython3.8.so (found version "3.8.5") found components:  Development 
-- Found ZLIB: /usr/lib/x86_64-linux-gnu/libz.so (found version "1.2.11") 
CMake Error at src/CMakeLists.txt:107 (find_package):
  By not providing "Findspdlog.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "spdlog", but
  CMake did not find one.

  Could not find a package configuration file provided by "spdlog" with any
  of the following names:

    spdlogConfig.cmake
    spdlog-config.cmake

  Add the installation prefix of "spdlog" to CMAKE_PREFIX_PATH or set
  "spdlog_DIR" to a directory containing one of the above files.  If "spdlog"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!

```
Not sure what should I do with it.

maliberty, 2021-05-25:
Clearly boost didn't get installed so that's why cmake fails.  

I tried the steps manually and I see something is odd.
    % wget https://github.com/boostorg/boost/archive/refs/tags/boost-1.72.0.tar.gz -o boost-1.72.0
   % tar -xf boost-1.72.0.tar.gz
   % ls
boost-1.72.0  boost-1.72.0.tar.gz  **boost-boost-1.72.0/**

The subdirectory is boost-boost-1.72.0 not boost-1.72.0 which is why cd fails.  @vvbandeira any ideas what has changed?  I assume you ran this to build the docker image.

vvbandeira, 2021-05-25:
I think this could be related to the recent change to where we download the tar file. I will take a look.

vvbandeira, 2021-05-26:
@Jenny-Tan can you try again with latest master branch? I just merged a possible fix with #676

Jenny-Tan, 2021-05-27:
I just did it. The boost library is now found but there are still errors.
```
-- The CXX compiler identification is GNU 9.3.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: 1
-- OpenROAD git sha: 223c78e86234fd3671c600c87187632ca1cf83a7
-- System name: Linux
-- Compiler: GNU 9.3.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- TCL readline library: /usr/lib/x86_64-linux-gnu/libtclreadline.so
-- TCL readline header: /usr/include/x86_64-linux-gnu
-- Found SWIG: /usr/bin/swig (found suitable version "4.0.1", minimum required is "3.0") 
-- Boost  found.
-- Found Python3: /usr/lib/x86_64-linux-gnu/libpython3.8.so (found version "3.8.5") found components:  Development 
-- Found ZLIB: /usr/lib/x86_64-linux-gnu/libz.so (found version "1.2.11") 
CMake Error at src/CMakeLists.txt:107 (find_package):
  By not providing "Findspdlog.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "spdlog", but
  CMake did not find one.

  Could not find a package configuration file provided by "spdlog" with any
  of the following names:

    spdlogConfig.cmake
    spdlog-config.cmake

  Add the installation prefix of "spdlog" to CMAKE_PREFIX_PATH or set
  "spdlog_DIR" to a directory containing one of the above files.  If "spdlog"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
```
Should I manually install spdlog?

maliberty, 2021-05-27:
Yes try it manually.  In general it would be helpful to have a log of when you ran DependencyInstaller.sh so we could see what happened.

vvbandeira, 2021-07-24:
@Jenny-Tan were you able to install and build? If so, please close the issue.

vvbandeira, 2021-09-06:
Closing due to inactivity, also the initial issue was fixed.

closed 2021-09-06 by vvbandeira