The-OpenROAD-Project/OpenROAD #632: Unable to build OpenRoad

Raynchowkw, 2021-05-10:
### Install dependencies
`sudo etc/DependencyInstaller.sh -dev` to install dependencies.
There occur some errors maybe because I downloaded dependencies manually several versions ago according to error messages and Dockerfile line by line.
The error message is:
```
epel-release-latest-7.noarch.rpm                         |  15 kB     00:00     
Examining /var/tmp/yum-root-WuL8ip/epel-release-latest-7.noarch.rpm: epel-release-7-13.noarch
/var/tmp/yum-root-WuL8ip/epel-release-latest-7.noarch.rpm: does not update installed package.
Error: Nothing to do
```

---
### Build by hand
`cmake`.  Error message:
```
-- Looking for C++ include pthread.h
CMake Error in /home/rayncc/openroad/OpenROAD/build/CMakeFiles/CMakeTmp/CMakeLists.txt:
  Target "cmTC_67abb" requires the language dialect "CXX17" , but CMake does
  not know the compile flags to use to enable it.

CMake Error at /usr/local/share/cmake-3.14/Modules/CheckIncludeFileCXX.cmake:92 (try_compile):
  Failed to generate test project build system.
Call Stack (most recent call first):
  /usr/local/share/cmake-3.14/Modules/FindThreads.cmake:110 (CHECK_INCLUDE_FILE_CXX)
  /usr/local/lib64/cmake/spdlog/spdlogConfig.cmake:4 (find_package)
  src/CMakeLists.txt:107 (find_package)

-- Configuring incomplete, errors occurred!
```

---
### Build using support using script
`$ ./etc/Build.sh`. Error message:
```
-- TCL readline library: /usr/lib64/libtclreadline-2.1.0.so
-- Boost  found.
CMake Error at /usr/local/share/cmake-3.14/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find Threads (missing: Threads_FOUND)
Call Stack (most recent call first):
  /usr/local/share/cmake-3.14/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  /usr/local/share/cmake-3.14/Modules/FindThreads.cmake:191 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  /usr/local/lib64/cmake/spdlog/spdlogConfig.cmake:4 (find_package)
  src/CMakeLists.txt:107 (find_package)

-- Configuring incomplete, errors occurred!
```

Could anyone have some solutions for these?
Thanks.

Raynchowkw, 2021-05-11:
Yes, Centos 7. I will try soon.

Raynchowkw, 2021-05-11:
In which step I should do these "source" and in which directory?
I guess I should do this just before "cmake" to get the compiler versions, right?
How to verify I "source" successfully?
When I do the first "source", it runs without giving any message,
but when I do the second, it says: "Can't read /etc/scl/conf/llvm-toolset-7.0, llvm-toolset-7.0 is probably not installed."
I guess this may be because I haven't installed dependencies correctly. I will find that in dockerfile.
But do you have some idea on what's going on with my "Install dependencies" process? How to verify if I install dependencies correctly?
Thanks.

Raynchowkw, 2021-05-11:
Thanks.
I check g++ and clang++ versions and they are correct.
To solve the llvm problem, I find it in the denpencyInstaller.sh.
I run the whole "_installCentosDev() {" part manually.
And then I run "`$ cmake`" or `$ ./etc/Build.sh` again.
The "compile flags" problem has been gone, but they are both stuck at
```
CMake Error at /usr/local/share/cmake-3.14/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find Threads (missing: Threads_FOUND)
```
which is the same as the result of `$ ./etc/Build.sh` I ran before.
What does "missing threads" mean?

Raynchowkw, 2021-05-11:
Yes. I can see it.

maliberty, 2021-05-11:
I looked at FindThreads.cmake and it tries to test compile:

#include <pthread.h>

static void* test_func(void* data)
{
  return data;
}

int main(void)
{
  pthread_t thread;
  pthread_create(&thread, NULL, test_func, NULL);
  pthread_detach(thread);
  pthread_cancel(thread);
  pthread_join(thread, NULL);
  pthread_atfork(NULL, NULL, NULL);
  pthread_exit(NULL);

  return 0;
}

Can you try that with your compiler and see what message you get?  You'll need to use -pthread when you compile & link

maliberty, 2021-05-11:
I put the above in bar.cc and ran:

g++ -o bar -pthread bar.cc

maliberty, 2021-05-11:
Then you have threads installed ok.  Would you try removing tools/OpenROAD/build and rebuilding with the correct compiler set.  Maybe there is some stale data.

Raynchowkw, 2021-05-11:
The build script can run to the end now. It says:
[100%] Built target trTest
But at the beginning, among lines of "detecting ...", "found ...", there is a line:
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS) 
Does it matter?

maliberty, 2021-05-11:
No, vtune is optional and can be ignored for your purposes (it is for runtime profiling).  I think you have a good build :)

closed 2021-05-12 by maliberty

Raynchowkw, 2021-05-18:
Hi @maliberty ,
I want to add 
```
source scl_source enable devtoolset-8
source scl_source enable llvm-toolset-7.0
```
so that I don't need to source these compilers every time.
in my bashrc, but I cannot find it in Centos 7.
Do you know what's the equivalent of bashrc in Centos 7?
Thanks.

maliberty, 2021-05-18:
It .bashrc and it will be the same on every platform.  I think you are missing the dot.

Raynchowkw, 2021-05-18:
Thanks. I've got it. It's ~/.bashrc.
I know the . before bashrc, but I just didn't know its location.