The-OpenROAD-Project/OpenROAD #633: PdnGen errors when stdcell power name is different than "VDD"

quentinw91, 2021-05-11:
Hi,
When I try to make floorplan on my platform, ```pdn.tcl``` crashes with a weird empty array error:
```
bash-4.2$ openroad -no_init -exit pdn.tcl 
OpenROAD 1 GITDIR-NOTFOUND
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0222] Reading LEF file: platforms/test_plat/lef/NangateOpenCellLibrary.tech.lef
[INFO ODB-0223]     Created 22 technology layers
[INFO ODB-0224]     Created 27 technology vias
[INFO ODB-0226] Finished LEF file:  platforms/test_plat/lef/NangateOpenCellLibrary.tech.lef
[INFO ODB-0222] Reading LEF file: platforms/test_plat/lef/NangateOpenCellLibrary.macro.mod.lef
[INFO ODB-0225]     Created 134 library cells
[INFO ODB-0226] Finished LEF file:  platforms/test_plat/lef/NangateOpenCellLibrary.macro.mod.lef
[INFO ODB-0127] Reading DEF file: results/2_5_floorplan_tapcell.def
[INFO ODB-0128] Design: gcd
[INFO ODB-0130]     Created 54 pins.
[INFO ODB-0131]     Created 457 components and 2057 component-terminals.
[INFO ODB-0133]     Created 414 nets and 1143 connections.
[INFO ODB-0134] Finished DEF file: results/2_5_floorplan_tapcell.def
[INFO PDN-0016] Power Delivery Network Generator: Generating PDN
  config: platforms/test_plat/pdn.cfg
[INFO PDN-0008] Design Name is gcd
[INFO PDN-0009] Reading technology data
[WARNING PDN-0018] No macro grid specifications found - no straps added
[INFO PDN-0011] ****** INFO ******
Type: stdcell, grid
    Stdcell Rails
      Layer: metal1  -  width: 0.170  pitch: 2.400  offset: 0.000 
    Straps
      Layer: metal4  -  width: 0.480  pitch: 56.000  offset: 2.000 
      Layer: metal7  -  width: 1.400  pitch: 40.000  offset: 2.000 
    Connect: {metal1 metal4} {metal4 metal7}
[INFO PDN-0012] **** END INFO ****
[INFO PDN-0013] Inserting stdcell grid - grid
-code 1 -level 0 -errorcode NONE -errorinfo {can't read "stripe_locs(metal4,POWER)": no such element in array
    while executing
"::odb::subtractSet $stripe_locs($layer_name,$tag) [dict get $blockages $layer_name]"
    (procedure "cut_blocked_areas" line 12)
    invoked from within
"cut_blocked_areas $tag"
    (procedure "add_grid" line 60)
    invoked from within
"add_grid"
    (procedure "plan_grid" line 46)
    invoked from within
"plan_grid"
    (procedure "apply" line 5)
    invoked from within
"apply $config"
    (procedure "pdngen::apply_pdn" line 13)
    invoked from within
"pdngen::apply_pdn $config_file $verbose "} -errorline 9
[ERROR PDN-9999] Unexpected error: can't read "stripe_locs(metal4,POWER)": no such element in array
Error: pdn.tcl, 11 PDN-9999
```

This looks to be coming from the fact that the power pins of the stdcell are different than "VDD" and "VSS"...

TESTCASE:
To reproduce it, I created a test platform (based on the nangate45, but without any macros).  In the _OK testcase, the ```pdn.tcl``` runs fine, while it crashes on the _NOK testcase.  The only difference between the 2 are the power pins defined in the platform are named "VDD_TEST" instead of "VDD" (same for VSS)
[20210511_OK.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/6459158/20210511_OK.zip)

[20210511_NOK.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/6459160/20210511_NOK.zip)

For both testcases, I ran the command ```openroad -no_init -exit pdn.tcl ```

quentinw91, 2021-06-21:
Hi,
any updates on this ?

maliberty, 2021-06-21:
@Colin-Holehouse @ibrahimkhairy @osamahammad21 please update with the current status.

osamahammad21, 2021-06-21:
@quentinw91 You have to update pdn::global_connections in the pdn.cfg as well with the new names:
set pdngen::global_connections {
    VDD_TEST {
        {inst_name .* pin_name VDD_TEST}
    }
    VSS_TEST {
        {inst_name .* pin_name VSS_TEST}
    }
}

Note: Since this is a core design  you need to add the pins layer in pdngen::specify_grid

pdngen::specify_grid stdcell {
    name grid
    rails {
        metal1 {width 0.17 pitch  2.4 offset 0}
    }
    straps {
        metal4 {width 0.48 pitch 56.0 offset 2}
        metal7 {width 1.40 pitch 40.0 offset 2}
    }
    connect {{metal1 metal4} {metal4 metal7}}
    pins { metal7 } #power pins on metal7
}

quentinw91, 2021-06-22:
Thanks, that fixed the pdngen issue.
But I now face a segmentation fault at the next step (global routing)
```
bash-4.2$ openroad -no_init -exit run_me.tcl 
OpenROAD 1 GITDIR-NOTFOUND
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0222] Reading LEF file: platforms/test_plat/lef/NangateOpenCellLibrary.tech.lef
[INFO ODB-0223]     Created 22 technology layers
[INFO ODB-0224]     Created 27 technology vias
[INFO ODB-0226] Finished LEF file:  platforms/test_plat/lef/NangateOpenCellLibrary.tech.lef
[INFO ODB-0222] Reading LEF file: platforms/test_plat/lef/NangateOpenCellLibrary.macro.mod.lef
[INFO ODB-0225]     Created 134 library cells
[INFO ODB-0226] Finished LEF file:  platforms/test_plat/lef/NangateOpenCellLibrary.macro.mod.lef
[INFO ODB-0127] Reading DEF file: results/2_floorplan.def
[INFO ODB-0128] Design: gcd
[INFO ODB-0130]     Created 56 pins.
[INFO ODB-0131]     Created 458 components and 2061 component-terminals.
[INFO ODB-0132]     Created 2 special nets and 916 connections.
[INFO ODB-0133]     Created 415 nets and 1145 connections.
[INFO ODB-0134] Finished DEF file: results/2_floorplan.def
[INFO GRT-0020] Min routing layer: metal2
[INFO GRT-0021] Max routing layer: metal10
[INFO GRT-0022] Global adjustment: 0%
[INFO GRT-0023] Grid origin: (0, 0)
[WARNING GRT-0043] No OR_DEFAULT vias defined.
[INFO GRT-0224] Chose via via1_4 as default.
[INFO GRT-0224] Chose via via2_8 as default.
[INFO GRT-0224] Chose via via3_2 as default.
[INFO GRT-0224] Chose via via4_0 as default.
[INFO GRT-0224] Chose via via5_0 as default.
[INFO GRT-0224] Chose via via6_0 as default.
[INFO GRT-0224] Chose via via7_0 as default.
[INFO GRT-0224] Chose via via8_0 as default.
[INFO GRT-0224] Chose via via9_0 as default.
[INFO GRT-0088] Layer metal1  Track-Pitch = 0.1400  line-2-Via Pitch: 0.1350
[INFO GRT-0088] Layer metal2  Track-Pitch = 0.1900  line-2-Via Pitch: 0.1400
[INFO GRT-0088] Layer metal3  Track-Pitch = 0.1400  line-2-Via Pitch: 0.1400
[INFO GRT-0088] Layer metal4  Track-Pitch = 0.2800  line-2-Via Pitch: 0.2800
[INFO GRT-0088] Layer metal5  Track-Pitch = 0.2800  line-2-Via Pitch: 0.2800
[INFO GRT-0088] Layer metal6  Track-Pitch = 0.2800  line-2-Via Pitch: 0.2800
[INFO GRT-0088] Layer metal7  Track-Pitch = 0.8000  line-2-Via Pitch: 0.8000
[INFO GRT-0088] Layer metal8  Track-Pitch = 0.8000  line-2-Via Pitch: 0.8000
[INFO GRT-0088] Layer metal9  Track-Pitch = 1.6000  line-2-Via Pitch: 1.6000
[INFO GRT-0088] Layer metal10 Track-Pitch = 1.6000  line-2-Via Pitch: 1.6000
[WARNING GRT-0039] Found pin outside die area in instance _348_.
[WARNING GRT-0039] Found pin outside die area in instance _348_.
...
[INFO GRT-0124] Update congestion history type 4.
[INFO GRT-0102] iteration 53, enlarge 23, costheight 529, threshold 0 via cost 0. 
log_coef 0.43634596, healingTrigger 30 cost_step 5 L 1 cost_type 1 updatetype 4.
[INFO GRT-0126] Overflow Report:
[INFO GRT-0127] total Usage   : 2731
[INFO GRT-0128] Max H Overflow: 41
[INFO GRT-0129] Max V Overflow: 29
[INFO GRT-0130] Max Overflow  : 41
[INFO GRT-0131] Num Overflow e: 11
[INFO GRT-0132] H   Overflow  : 55
[INFO GRT-0133] V   Overflow  : 44
[INFO GRT-0134] Final Overflow: 99

[INFO GRT-0124] Update congestion history type 4.
[INFO GRT-0102] iteration 54, enlarge 23, costheight 539, threshold 0 via cost 0. 
log_coef 0.43634596, healingTrigger 31 cost_step 5 L 1 cost_type 1 updatetype 4.
[WARNING GRT-0164] initial grid wrong y1 x1 [0 0] , net start [0 1] routelen 54.
[WARNING GRT-0167] checking failed 1.
Segmentation fault (core dumped)
```

Looks like the possibility to have IO placement (and not just core design) modified the way to setup platforms, but it is not clear to me what was changed.   I was able with previous versions of OpenRoad to create a new platform by simply looking at what was done in nangate45, but this does not work anymore... Is there an example where the platform is only setup for core design ? 

Anyway, here is the testcase with the segmentation fault:
[20210622.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/6693481/20210622.zip)

maliberty, 2021-06-22:
Please don't tack entirely new issues on to this report.  Please open a new issue for global route.  This issue is resolved by your reporting.

closed 2021-06-22 by maliberty

quentinw91, 2021-06-22:
Hi, 
Sorry, it was difficult for me to see if the issue that i am facing is related or not to the global connection definition.  So difficult to know if it is a follow-up of the fix proposed in this one or something else.  But no problem, I will open a new issue...