The-OpenROAD-Project/OpenROAD #634: Documentation for reports generated by check_antennas

ahmednofal, 2021-05-11:
Thank you for your work on this project. The antenna violations reported by `check_antennas` command are dumped in a report file. It is not clear what the syntax of that file is and how to read it.

Can you please offer some documentation or guidance for that ?

ahmednofal, 2021-05-24:
Hello, thank you for the response, any progress in this regard, should I close the issue?

maliberty, 2021-05-24:
I know @msaligane is busy with a tapeout at the moment so it might be a bit until he responds.  If you have a specific question I can try to help.

msaligane, 2021-05-24:
Sorry for the delay. 
@maliberty where should I add the doc for this?

maliberty, 2021-05-24:
You can make a src/ant/doc directory to hold your module docs.  Please add a link to it from the top-level README.

donn, 2021-06-15:
You never ended up committing that documentation file. We're kind of blocked by this. Any ETA?

maliberty, 2021-06-15:
@msaligane please give an eta

msaligane, 2021-06-15:
I will work on this tonight. Apologies I have been busy with the tapeout and other misc. things.

msaligane, 2021-06-17:
@ahmednofal can you be more specific on what is unclear on the report? I can maybe add some explanations if the report is unclear.

donn, 2021-06-18:
How to interpret the report at all. There are many symbols and abbreviations that do not appear to be standard. All the report can successfully tell us now is that there are violations, the data without context is not particularly useful.

maliberty, 2021-06-18:
Mehdi is working on a document but I suggest you look at the lef/def manual as it is the basis for all the checking and has a lot of explanation of the terms.

msaligane, 2021-06-19:
I added some more documentation.  You will need to dig a little more to understand how the ratios are calculated. I added a simple example of how it is done (step-by-step).

donn, 2021-06-22:
https://openroad.readthedocs.io/en/latest/search.html?q=antenna&check_keywords=yes&area=default

closed 2021-06-24 by msaligane